canboat/canboat #28: n2kd crashes when connecting to caching server

stevepmitchell, 2015-01-11:
Using the most recent version of canboat.
Have an Actisense serial NGT-1, not USB.

Running "actisense-serial /dev/ttyS1 | analyzer  -json | n2kd" to try to debug why n2kd_monitor was restarting constantly.

When connecting to the caching server on port 2597, n2kd exits with the below.  This can be reproduced every time I try to connect to the caching server.  I suspect it's a device on my network (very busy) that is throwing crappy data, but I can't be 100% sure:

FATAL 2015-01-11 11:07:01.504 [n2kd] Stream 0 contains invalid fd 976629818

Connecting to the non-caching json server on 2598, and the NMEA0183 server on 2599 work fine for long periods of time.

canboat, 2015-01-25:
Can you check whether the latest commit https://github.com/canboat/canboat/commit/d0a4c4a2109e34dbda48e3b51202875c7e45bcf8 fixes this? I couldn't find the issue exactly, but there was some fishy stuff going on in closeStream. Also, if it does still happen, run n2kd in debug mode and check the output.

stevepmitchell, 2015-01-25:
Well things are more stable with the checkin above, but now port 2597 is completely blank when I request any data from it.  I get a TCP connect, and then the session is closed.

canboat, 2015-01-27:
@stevepmitchell did commit 8cdad67ce457fc986fc764812a858f88130a8d2c fix it? That cleared up some remaining issues that I found. Port 2597 worked for me in that version.

canboat, 2015-04-01:
Assuming this worked as I received no further feedback.

closed 2015-04-01 by canboat