canboat/canboat #31: unit conversion MWV msg

villalolo, 2015-01-26:
Hello,

I have made the following change to transform m/s (n2k) to knots (N) in the 183 msg in the wind speed.

Line 235
   float m2n;

Line 249

  m2n = strtof(speed, NULL);
  m2n = m2n * 1.944;
  snprintf(speed,12, "%2.1f", m2n);
.... 
if (strcmp(reference, "True") == 0)
  {
    
	nmea0183CreateMessage(msg183, src, "MWV,%s,T,%s,N,A", angle, speed);

canboat, 2015-01-27:
Hi @villalolo, first hurdle done -- you're creating issues.

You're right the units should be translated from m/s to the unit indicated in the NMEA0183 sentence. As you can see this nmea0183 code is still a bit of a hack that isn't well tested.

closed 2015-01-27 by canboat

villalolo, 2015-01-27:
what you guys are doing is amazing ...

commit 5a47b4c0 (2015-01-27): Add speed conversion to NMEA0183 MWV sentence.

Fix #31.

The code now produces normalized values in KM/h.