canboat/canboat #33: Saturation of the nma 183 network

villalolo, 2015-01-27:
Is it possible to limit the number message going out in time. the network 183 is working at 4800 bds, and saturates very easily. 
$WIMWV,156.8,R,016.08,N,A*08, is sent 10 time/sec = 32 x 10 x 10 = 3200 bds
For instance, a GPS can send once per second.

canboat, 2015-01-27:
This is TCP/IP so where is the 4800 baud limit coming from?

canboat, 2015-01-27:
Are you stuffing this data back onto a serial port into an actual NMEA0183 network?

villalolo, 2015-01-27:
The nmea is 4800, I feed MWV to an autopilot, but it also receive data from GPS and Navsoftare (way points)

villalolo, 2015-01-27:
The Message is going from RPi to Arduino who dispatch and filter then to serial port

canboat, 2015-01-27:
How's that for turn around time?

closed 2015-01-27 by canboat

villalolo, 2015-01-27:
I can see 2 ways, either drop everything that comes up too early, either drop m-1 out of m msg...

canboat, 2015-01-27:
I already put in a fix: if you set `--rate-limit` on the `n2kd` command-line it will only report a particular message type once a second.

villalolo, 2015-01-27:
Man... this is great ! seriously ...

canboat, 2015-01-27:
I've been putting off sending more NMEA0183 sentences, would it help you if it also sent out a GPS location?

villalolo, 2015-01-27:
Personnaly, I only have one n2k instrument..... the wind data  ..... but I think the more NMEA message that program can do the best.

commit 2591a610 (2015-01-27): Add rate-limit option to NMEA0183 stream.

If --rate-limit is set every sender / PGN combo will only be reported once per second on the NMEA0183 stream.
This is useful for users that intend to push the stream onto a physical NMEA0183 connection at 4800 baud.

Fix #33.