canboat/canboat #34: candump2analyzer works only with a specific candump output formatting.

woobagooba, 2015-03-05:
Testing candump2analyzer on several linux platforms reveals candump formatting differs.

Modify candump2analyzer to be more tolerant of variations in the candump format.

woobagooba, 2015-03-05:
Could not assign this to myself, but I'll take the action item.

alexphredorg, 2015-03-31:
I wrote a python script that uses socketcan to read the CAN bus natively.  I'm doing my own NMEA 2000 parsing (using pgns.json to translate the data to something useful), but it would be easy to make this into a front end for analyzer if you want a copy of it.

That is probably safer than try to parse different flavors of candump.

woobagooba, 2015-04-01:
IMO more useful if we had raw socket bytes -> analyzer input format?  Then pipe into the analyzer.

alexphredorg, 2015-04-01:
I'm happy to do that.  Is the time actually used by anything?  If not, why is it there?

tkurki, 2015-04-01:
You mean the timestamp? It is useful at least for time-accurate playback of log files.

canboat, 2015-04-01:
The Github way of providing improvements is forking the entire project, implement your improvement in your own tree then send me a pull request.

canboat, 2015-04-01:
Ah I forgot, you already did. Didn't your commit https://github.com/canboat/canboat/commit/b1fc35b52b5536b6c9b10dcf4da49b4eb5f6e50a fix this issue?

closed 2018-07-24 by keesverruijt