canboat/canboat #44: Fallback mechanism for "unknown" PGN entries is unclear

acolomb, 2015-07-03:
Could someone with a thorough understanding of the code please document what ```analyzer``` does when finding an unknown PGN?  There are several entries with the ```unknownPgn``` field set to true.  Any unknown PGN seems to fall back to the "unknown" entry with the highest number less than or equal to its own.  However I recall some instances where the fallback went wrong.

I tried to understand what the code does, but it's rather hard to follow.  The first step to verify if it's working alright would be to have a clear documentation about the expected order in ```pgn.h```.

* Should known PGN entries always stand before or after the catch-all ones?
* Which specific catch-all entry gets used?
* Is there a difference in behavior whether an entry was found with the correct PGN but mismatched fixed fields, or no matching PGN entry at all?

acolomb, 2015-07-03:
For some background... Especially proprietary PGNs could have several possible fall-back entries:

For example, from most generic to most specific:
* proprietary PDU1 single-frame: Manufacturer ID and Industry Code can be decoded.
* manufacturer specific PDU1 single-frame (fixed Manufacturer ID, unknown Proprietary ID): proprietary fields with a default structure may be decoded in addition (such as a Device Instance field always at the same position)
* manufacturer specific PGN sub-instance distinguished by an ID field (fixed Manufacturer ID, fixed Proprietary ID): all fields may be decoded

keesverruijt, 2022-10-17:
This has been fixed in v3.0.0, with explicit fallbacks for each range, with the ranges explained in the documentation page generated from the new `docs/canboat.xml` file using `docs/canboat.xsl` available at https://canboat.github.io/canboat

closed 2022-10-17 by keesverruijt