canboat/canboat #48: analyzer crashes when multiple 129029

dfrosch, 2015-08-28:
Working on a n2k reader that reads raw CAN bus, I tried to inject these messages :

2015-08-28T14:06:27.614Z,0,129029,0,255,8,01,ff,ff,ff,ff,ff,ff,ff
2015-08-28T14:06:27.614Z,0,129029,0,255,8,02,ff,7f,ff,ff,ff,ff,ff
2015-08-28T14:06:27.614Z,0,129029,0,255,8,03,ff,ff,7f,ff,ff,ff,ff
2015-08-28T14:06:27.614Z,0,129029,0,255,8,04,ff,ff,ff,7f,03,fc,00
2015-08-28T14:06:27.614Z,0,129029,0,255,8,05,ff,7f,ff,7f,ff,ff,ff
2015-08-28T14:06:27.614Z,0,129029,0,255,8,06,7f,00,ff,ff,ff,ff,ff
2015-08-28T14:06:27.614Z,0,129540,0,255,8,00,03,ff,ff,00,ff,ff,ff

analyzer crashes :
#7  0x0804d82f in printPacket (index=135, unknownIndex=74, msg=msg@entry=0xbfea02bc) at analyzer.c:2006
#8  0x0804da5e in printCanFormat (msg=msg@entry=0xbfea02bc) at analyzer.c:2153
#9  0x08049507 in main (argc=2, argv=0xbfea0c44) at analyzer.c:461

Is analyzer doing Fast Packet reassembly or should I do it before ?

Thx for this excellent project !

canboat, 2016-01-23:
Ah, your PGN is wrongly formatted as it doesn't contain a length in byte 6 of the first fast packet part. But of course the code should just accept the (wrong) length of 255 bytes for the PGN.

canboat, 2016-01-24:
Commit https://github.com/canboat/canboat/commit/21c4746fd3c3dbb59e6221e72617c04b30178f2a fixes this issue.

closed 2016-01-24 by canboat