canboat/canboat #50: Nonstandard timestamp

tkurki, 2015-11-02:
Json timestamps look like 
2013-10-08-15:47:28.263 and afaik should be 
2013-10-08T15:47:28.263Z

canboat, 2016-01-23:
Fixed by commit 60b98501895a46700ba7889b87764340e74e4693

closed 2016-01-23 by canboat

tkurki, 2016-02-02:
```
~/git-workspace/canboat (master) $ make clean
...
~/git-workspace/canboat (master) $ make
...
cc -O3 -o ../rel/darwin-x86_64/analyzer -I../common analyzer.c ../common/common.c 
~/git-workspace/canboat (master) $ cat ../signalk/server-node/samples/n2k-1hour.data | rel/darwin-x86_64/analyzer -json | head -1
...
{"timestamp":"2014-08-15-19:00:00.042","prio":3,"src":160,"dst":255,"pgn":129029,"description":"GNSS Position Data","fields":{"SID":135,"Date":"2014.08.15", "Time": "18:59:59.04620","Latitude":59.7250108,"Longitude":24.7366770,"GNSS type":"GPS","Method":"GNSS fix","Integrity":"No integrity checking","Number of SVs":10,"HDOP":0.80,"Geoidal Separation":-0.01,"Reference Station ID":7}}
```

tkurki, 2016-02-02:
My mistake, it is copied from the sample file verbatim.