canboat/canboat #55: 129540 Satellites in view: use array?

tkurki, 2016-02-14:
JSON output for 129540 looks like n times
```
    "PRN XX": "67",
    "Elevation XX": "41.0",
    "Azimuth XX": "207.0",
    "SNR XX": "31.00",
    "Range residuals XX": "0",
    "Status XX": "Used"
```

How about making the satellites' details an array?

tkurki, 2016-02-14:
Related: https://github.com/SignalK/n2k-signalk/issues/12

tkurki, 2016-02-14:
See also https://github.com/SignalK/specification/issues/163#issuecomment-183304174

closed 2016-02-15 by canboat

tkurki, 2016-02-15:
Almost: the array is a property and needs a name: 

```
  "fields": {
    "SID": 207,
    "Sats in View": 11,
    [{
      "PRN": 32,
      "Elevation": 69.0,
      "Azimuth": 223.0,
      "SNR": 32.00,
```

canboat, 2016-02-15:
Do you think I should put a name for the array in the datamodel so that it says `"satellites":` for this PGN or can I get away with `"list": ` ?

canboat, 2016-02-15:
(or something else instead of list?)

commit cc85c963 (2016-02-15): Fix #55

Repeating fields are now output as an array, for all applicable PGNs.
This is a breaking change to the JSON output.