canboat/canboat #56: Invalid pngs.json

jeronimo, 2016-03-03:
Not sure how this is generated but in **pngs.json** file for **127489** png for `discreteStatus1` and `discreteStatus2` fields **EnumBitValues** is invalid JSON (from https://en.wikipedia.org/wiki/JSON - Object: an unordered collection of name/value pairs where the names (also called keys) are strings) and can't be parsed by other parsers.
Instead of `{Bit:'0',Name:'Check Engine'}` shoud be `{'Bit':'0','Name':'Check Engine'}`

If there won't be any more xml changes I can just update that json file manually and make pull request myself.

tkurki, 2016-03-04:
See https://github.com/canboat/canboat/blob/master/analyzer/Makefile#L34.

It used to work. Imho it would be a better idea to fix the generation, not the result. 

An optional json validation step would not hurt, maybe with vanilla ruby or python so that you would not need to install extra tools for that.

canboat, 2016-03-19:
'them bitvalues, they be new yah!

commit eccdab29 (2016-03-19): Fix #56 - Generate proper JSON from XML for EnumBitPair
commit 2eaad67d (2016-03-22): Merge pull request #60 from jeronimo/master

Fix #56 - Generate proper proper JSON from XML for EnumBitPair
commit 35c0efb8 (2016-03-22): Fix #56 - Generate proper proper JSON from XML for EnumBitPair
closed 2016-03-19 by canboat