canboat/canboat #63: Reading from can0 interface

Mehul2802, 2016-05-13:
Hi,

actisense serial module reads from either serial or USB interface on Linux platform. I have one board which exports can0 interface. So, we can read-write to this interface via SocketCAN API.

I want to read-write date from can0 network interface
Is this method supported in current release by any way? 

-Mehul

canboat, 2016-05-13:
Yes. The documentation can be improved, but you can use the standard linux command 'candump' with https://github.com/canboat/canboat/tree/master/candump2analyzer for reading N2K messages as well as https://github.com/canboat/canboat/tree/master/socketcan-writer for writing N2K messages.

chacal, 2016-05-13:
Hi,

Yes, it is possible by using commonly available `canutils` package and canboat together.

To read data from can0 create the following pipeline:

`candump can0 | candump2analyzer | analyzer`

To write to can0 `socketcan-writer` tool can be used. For example to request product information using PGN 59904:

`echo "1970-01-01-00:00:00.000,6,59904,0,255,3,14,f0,01" | socketcan-writer can0`

Mehul2802, 2016-05-13:
Ok. By using candump, I will read data from can0 and then pass it to candump2analyzer. So, It will convert the data to analyzer's plain text format.

Thanks @chacal @canboat

closed 2016-05-13 by Mehul2802