canboat/canboat #65: Small size of can dump on vcan0 is not getting forwarded by n2kd to TCP clients

Mehul2802, 2016-05-26:
Hi,

I am using below command to send data over TCP.
$ candump vcan0 | ./candump2analyzer | ./analyzer | ./n2kd

I am using another application which sends data on vcan0.

I am able to fetch data on TCP client side when data is above 7kB.
However, when I send less than 7kB or just one packet on vcan0 then I am not getting data on TCP client side.

Regards,
Mehul

canboat, 2016-05-26:
Something is buffering. Check if adding `fflush(outfile)` to candump2analyzer fixes this for you.

Mehul2802, 2016-05-27:
Hi,

By using fflush(outfile) in candump2analyzer, I am getting dump of one packet on TCP client.

Thanks @canboat 
Mehul

closed 2016-05-27 by Mehul2802

JohanElmis, 2016-07-22:
Hi,

I thought  you were interested in stuff like  that, so probably that article would be helpful to you,  please read it here  <http://kind.247oracle.com/e4pse>

johan@elmerfjord.com