canboat/canboat #66: NMEA0183 PGNs

stevepmitchell, 2016-05-27:
This is more of a feature request or a question. Is there any easy way or plan to add more PGN's to the NMEA0183 server? Specifically I am looking for the battery and other power monitoring ones as I want to use iNavX and other 3rd party tools to see that data.  SignalK and some of the other tools aren't fully supported yet by remote apps, and can't figure out another way to do this. 

Perhaps if I could somehow send the NMEA 2000 stream directly out some of these apps would understand?  Or am I missing some functionality that might already help me?

tkurki, 2016-05-28:
Signal K node server does provide nmea0183 over tcp for any nmea 0183 inputs. I am currently working on n2k->nmea0183 conversion to send to my tillerpilot & tablet apps. The same solution would probably serve you.

What specific conversions are you looking for?

stevepmitchell, 2016-05-28:
I totally missed that there is a NMEA 0183 server for SignalK.  Is it in a mode I could test against? I see a parser at https://github.com/SignalK/signalk-parser-nmea0183.

In particular, I'm looking to convert the following NMEA 2000 PGN's:

127505
127506
127507
127508

canboat, 2016-10-19:
Steve, enhancing the NMEA0183 support in N2KD is not high on my list, but I will accept pull requests.

closed 2016-10-28 by canboat