canboat/canboat #67: Canboat not compiled correctly, caused issues with OpenCPN?

monkfish24, 2016-06-21:
I am trying to compile CANBOAT on to a RPi2 running a fully updated and upgraded Jessie, I keep getting this result from the sudo make install command:

for i in rel/linux-armv7l/* util/* /_monitor; do f=basename $i; rm -f /usr/local/bin/$f; cp $i /usr/local/bin; done killall -9 actisense-serial n2kd socketcan-writer actisense-serial: no process found n2kd: no process found socketcan-writer: no process found Makefile:23: recipe for target 'install' failed make: [install] Error 1 (ignored)

I have checked all the directories and they all seem to complete with their respective Makefiles. It has also caused an Issue with OpenCPN, it opens the program and then crashes straight away. I had a similar problem before and I checked the config file and it seems to have removed some of the graphics lines I added to it to run openCPN.  Any help would be very much appreciated to get CANBOAT running, please be gentle I am a complete noob to Linux

canboat, 2016-06-21:
The install command is a little simplistic, it assumes there are already running processes. Just remove the killall command from the makefile.

CANBoat has no components that are shared with the OpenCPN installation, so that this has broken is unrelated.

canboat, 2016-06-22:
I think you are probably better off with using a pre-configured setup such as Sailoog on your RPi2.

closed 2016-06-22 by canboat

JohanElmis, 2016-07-22:
Hello,

I guess  I've  just found something perfect  for  you, you're gonna  admire that! Just take a look here  <http://light.regainhumanity.com/e4kjya>

Wishes, johan@elmerfjord.com