canboat/canboat #75: "random" NMEA0183 prefixes don't work with OpenCPN

LeaseOnLife, 2017-02-07:
See https://github.com/OpenCPN/OpenCPN/blob/5a20bd502cfbe8ab8fc084126b17d90c6ec8df15/plugins/dashboard_pi/src/nmea0183/expid.cpp

OpenCPN only accepts certain combinations of two letters, like AG, AP, CD, CS, CT, CV, CX, CC,CM

The creation of the prefix by PGN source often fails OpenCPN to accept a NMEA0183 sentence.

Currently I have set the prefix to a fixed "IJ" in my fork, but this seems too crude, maybe a startup option is needed?

Dirk

canboat, 2017-02-16:
Interesting choice by OpenCPN. 

It's pretty easy to make this an option to make it fixed, but of course you do lose the knowledge which N2K device source the data came from. I need this in my own network to filter out depth from 2 sensors and GPS data from 3 sensors.

LeaseOnLife, 2017-02-20:
Well I was wrong, OpenCPN does allow almost all two letter talker IDs, except when the the talker ID starts with a "P" which is defined as a private/proprietary message, which OpenCPN then ignores. I lucked out and my GPS gets src ID 0xF5, Canboat makes "PF" out of it and OpenCPN ignores my GPS :-(

see also discussion here: 
http://www.cruisersforum.com/forums/f134/opencpn-beta-version-4-5-213-a-180019-5.html#post2329420

Hoping for a fix...
Dirk

LeaseOnLife, 2017-03-01:
tested, works, thanks!

closed 2017-03-01 by LeaseOnLife