canboat/canboat #76: Issue with "make" function - installation of canboat

ndelaet, 2017-03-01:
Hi, 

Totally beginner with RPi, I would like to setup a SignalK server for my boat, using a Pi. At this stage, I don't know much about programming, but intend to end-up with capability to send my boat's data to the cloud, and remote-access the boat's systems from home.

Sadly, I ran into my first (of many) issues right when trying to install canboat utility on my new Pi 3B with last version of Jessie. Issue occured when executing the "make" command

`make[1]:`
 Entering directory '/home/pi/canboat/analyzer'
../rel/linux-armv7l/analyzer -explain-xml >pgns.xml && xsltproc pgns2json.xslt pgns.xml >pgns.json
/bin/sh: 1: xsltproc: not found
Makefile:34: recipe for target 'json' failed
make[1]: *** [json] Error 127
make[1]: Leaving directory '/home/pi/canboat/analyzer'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error ``

then, when running "make install", I got this:

`pi@raspberrypi:~/canboat` $ make install
for i in rel/linux-armv7l/* util/* */*_monitor; do f=`basename $i`; rm -f /usr/local/bin/$f; cp $i /usr/local/bin; done
cp: cannot create regular file ‘/usr/local/bin/actisense-serial’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/analyzer’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/candump2analyzer’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/command-group-function’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/iptee’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/n2kd’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/n2kd_monitor’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/nmea0183-serial’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/request-group-function’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/socketcan-writer’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/list-product-information’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/n2kd_monitor’: Permission denied
cp: cannot create regular file ‘/usr/local/bin/n2kd_nmea_monitor’: Permission denied
Makefile:24: recipe for target 'install' failed
make: *** [install] Error 1
``
Tried to remove canboat folder, and restart from the beginning. Same issue.
As I'm new, very sorry in advance for mis-formatting, or mis-manipulations.

Thanks in advance for your help

ndelaet, 2017-03-01:
Sorry, mis read explanation

closed 2017-03-01 by ndelaet

canboat, 2017-03-01:
Fixed the 'building' page in the Wiki. You need to install xsltproc and use `sudo` before `make install`.