canboat/canboat #77: Cannot start the n2kd_monitor (end of anboat install)

ndelaet, 2017-03-01:
Hi, 

After successful creation of the /etc/default/n2kd file in the right folder (struggled a bit, because rights are limited --> had to do it in "root" user mode), I now have an issue when trying to start the server:

`pi@raspberrypi:~ $ /usr/local/bin/n2kd_monitor
Can't locate ConfigReader/Simple.pm in @INC (you may need to install the ConfigReader::Simple module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.20.2 /usr/local/share/perl/5.20.2 /usr/lib/arm-linux-gnueabihf/perl5/5.20 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.20 /usr/share/perl/5.20 /usr/local/lib/site_perl .) at /usr/local/bin/n2kd_monitor line 38.
BEGIN failed--compilation aborted at /usr/local/bin/n2kd_monitor line 38.
pi@raspberrypi:~ $ 
`

I've looked for this ConfigReader, but cannot find reference on the net (?)
(sorry again if my questions are stupid ;))

thks

canboat, 2017-03-01:
'perl configreader' search in google would have resulted in the top hit being correct, but I've added the necessary installation step to the n2kd_monitor page in the wiki.

closed 2017-03-01 by canboat

ndelaet, 2017-03-01:
Thanks a lot,
Seems to work now. Next step for me: connect the Actisense NGT1 and make it work ;)
kr