canboat/canboat #78: Manufacturer Code as Code, not Name

tkurki, 2017-03-10:
`Manufacturer Code` in for example [PGN 60928 fields](https://github.com/canboat/canboat/blob/master/analyzer/pgn.h#L654) is being translated to a human oriented manufacturer name.

This is a bit confusing (`Code` is actually `Name`) and if a downstream app would like to have access to the original value there is no convenient way to do that.

I propose that Manufacturer Code should be output by default as the untranslated numeric value and there should be an option to additionally include the translated value as `Manufacture Name`field (`-manufacturerName`) and for backwards compatibility perhaps an option to function like it does today (`-manufacturerNameAsCode`).

Thoughts?

canboat, 2017-04-22:
Mja, mja, this is actually true for a lot of values where analyser translates values via a lookup table. The JSON should have been structured differently, really.

canboat, 2017-04-22:
Adding a new `-nv` (name/value) option to the `-json` output of analyser.

With `-json` the output is:

    {"timestamp":"2011-04-25-09:08:49.755","prio":3,"src":14,"dst":255,"pgn":65323,"description":"Manufacturer Proprietary single-frame non-addressed","fields":{ "Manufacturer Code": "Simrad","Industry Code":"Marine","Data":"281474959999231"}}

With `-json -nv` the output is:

    {"timestamp":"2011-04-25-09:08:49.755","prio":3,"src":14,"dst":255,"pgn":65323,"description":"Manufacturer Proprietary single-frame non-addressed","fields":{ "Manufacturer Code":{"value":1857,"name":"Simrad"},"Industry Code":{"value":4,"name":"Marine"},"Data":"281474959999231"}}

commit e3e4c9a2 (2017-04-22): Fix #78

Adding a new `-nv` (name/value) option to the `-json` output of analyser.

With `-json` the output is:

    {"timestamp":"2011-04-25-09:08:49.755","prio":3,"src":14,"dst":255,"pgn":65323,"description":"Manufacturer Proprietary single-frame non-addressed","fields":{ "Manufacturer Code": "Simrad","Industry Code":"Marine","Data":"281474959999231"}}

With `-json -nv` the output is:

    {"timestamp":"2011-04-25-09:08:49.755","prio":3,"src":14,"dst":255,"pgn":65323,"description":"Manufacturer Proprietary single-frame non-addressed","fields":{ "Manufacturer Code":{"value":1857,"name":"Simra [... cut]
closed 2017-04-22 by canboat