canboat/canboat #82: actisense-serial on armhf (raspbian jessie)

jlaba, 2017-04-23:
actisense-serial quits with a segfault on raspbian jessie. Is there a way to debug the programm examine the problem in more detail?

Regards,

Joachim

canboat, 2017-04-23:
Compile with debug mode, then examine the core dump in gdb.

jlaba, 2017-04-24:
I figured out how to create core dump files and with my very little knowledge of gdb I tried to find out the segfault reason.

In case of
 * n2kd it seems to be an issue in pgnIdx in _start
 * analyzer pgnList in _start
 * actistense_serial ?

I have attached the core dump files for those programs.

Interestingly, the segfault does not appear, if I compile canboat on raspbian itself. Compiling it on debian armhf leads to this segfaults. I don't observed this by other packages.

Do you have an Idea, what the problem might be?

canboat, 2017-04-24:
I’m very sympathetic, but as you realise probably it is very hard for someone who doesn’t have your hardware to reproduce this.

The fact that you need a cross deploy to trigger the bug indicates something is not alike between the 2 systems.

A core dump in _start seems to indicate that there is an incompatibility between the dynamic libraries of the 2 systems. If it really crashes in _start, it means the canboat code hasn’t even run yet.

jlaba, 2017-04-24:
Thanks for your reply. I awaited such a bad situation. I created an raspbian package to bypass it. The package could be found in http://ftp.engsas.de/unstable/gnuboat-raspbian/ repository.

closed 2017-04-24 by jlaba