canboat/canboat #84: n2kd servers should return HTTP compliant repsonse or not?

jlaba, 2017-04-25:
Currently n2kd just returns the json values. This may (and e.g. with python requests module it does) lead to problems consuming the page, using standard compliant clients.

So my question is, should n2kd servers return a HTTP compliant response or not? If yes, I'm try to implement it.

See http://www.w3.org/Protocols/rfc2616/rfc2616-sec6.html for what a compliant response should contain.

woobagooba, 2017-04-25:
Do you mean an HTTP compliant Response body, or a full RESTful interface?

HTTP/1.1: Response - World Wide Web Consortium<http://www.w3.org/Protocols/rfc2616/rfc2616-sec6.html>
www.w3.org
The first line of a Response message is the Status-Line, consisting of the protocol version followed by a numeric status code and its associated textual phrase, with ...

canboat, 2017-04-25:
n2kd server is a minimal C program that does NOT implement HTTP. At all. Period. It uses its own minimal protocol, which is suited to a frontend that implements HTTP.

I know RFC2616 (superceded by RFC7230) back-to-front and implementing it from scratch is a lot of work.

canboat, 2017-04-25:
If somebody wants to invest the time and write up a simple HTTP server, or a small wedge between (for instance) NGINX and n2kd I'd be more than happy to integrate your pull request :-)

closed 2017-04-25 by canboat

jlaba, 2017-04-25:
Maybe https://www.gnu.org/software/libmicrohttpd/ is a useful library to use as http server within this project. It seems to be very small, powerful enough and is implemented in C.

Might this be an option for you?

tkurki, 2017-04-25:
@jlaba are you aware of https://github.com/SignalK/signalk-server-node ?

Embeds canboat, data over http(s) and websocket and tcp, data from nmea2000 and nmea0183 in uniform format, conversions to nmea0183 etc.

jlaba, 2017-04-27:
Thanks for the hint!

Seems to be very interesting. As I understood Node Js 6 is needed and since my 
goal is to install everything from packages (therefore I packaged n2kd) I will 
have a deeper look at signalK after the first release of my project ;-)

tkurki, 2017-04-27:
See also https://github.com/tkurki/marinepi-provisioning

jlaba, 2017-04-28:
Your one step ahead ;-) But my puppet module is also ready now.