canboat/canboat #93: n2kd_monitor spawns additional n2kd_monitor processes

jarmund, 2017-08-31:
What the title says. After running n2kd_monitor once to set up the pipeline once, an additional n2kd_monitor process is spawned about twice per minute. During testing, I found this:

> root@navy:\~/n2k# ps aux | grep n2kd
> root      1617  0.0  0.2  25152  5164 pts/1    S    10:25   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> root      1633  0.0  0.1  13228  2828 pts/1    S    10:25   0:00 /bin/bash -c actisense-serial -t 10 /dev/ttyUSB0 | analyzer -json -json | n2kd 
> root      1636  0.7  0.0  40700  1272 pts/1    S    10:25   0:04 n2kd
> root      1654  0.0  0.1  25152  3828 pts/1    S    10:26   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> root      1675  0.0  0.1  25152  3652 pts/1    S    10:26   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> root      1685  0.0  0.1  25152  3652 pts/1    S    10:27   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> root      1691  0.0  0.1  25152  3652 pts/1    S    10:27   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> root      1692  0.0  0.1  25152  3652 pts/1    S    10:28   0:00 /usr/bin/perl /usr/local/bin/n2kd_monitor
> .....snip......
> root      1798  0.0  0.1  12728  2116 pts/0    S+   10:35   0:00 grep n2kd
> root@navy:~/n2k# pstree
> systemd─┬─acpid
>         ├─agetty
>         ├─atd
>         ├─cron
>         ├─dbus-daemon
>         ├─dhclient
>         ├─exim4
>         ├─n2kd_monitor─┬─bash─┬─actisense-seria
>         │              │      ├─analyzer
>         │              │      └─n2kd
>         │              └─n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor───n2kd_monitor──........etc

jarmund, 2017-08-31:
Dirty workaround: Comment out line 163 through 177, as I have other processes monitoring potential failure.

> #    if (!$monitor && (time > $last_monitor + 30))
> #    {
> #      $last_monitor = time;
> #      if (($monitor = fork()) == 0)
> #      {
> #        open STDIN, '/dev/null' or die "Can't read /dev/null: $!";
> #        open STDOUT, '>>', $MONITOR_LOGFILE or die "Can't write to $MONITOR_LOGFILE $!";
> #        open STDERR, '>&STDOUT' or die "Can't dup stdout: $!";
> #        exec 'php5', '/usr/local/bin/n2k.php', '-monitor';
> #      }
> #      if (!$monitor)
> #      {
> #        logText("Fork monitor failed.");
> #      }
> #    }

canboat, 2017-08-31:
Serves you right for using systemd :-)

I don't have time right now to see why the fork is not returning 0, or what else is going on, and I don't use systemd.

Can you investigate in more detail and preferably provide a pull request that works on both systemd and traditional/better rc systems?

Knobee, 2019-01-03:
This now lives around line 228 - 242 of n2kd_monitor  (and it just bit me (again), so it's still an issue)

Knobee, 2019-01-04:
On 1/3/19 11:26 AM, Kees Verruijt wrote:
> Feel free not to use n2kd_monitor at all, or send me a pull request.

That is my plan.

> Remarks like 'it just bit me' are fine but what do you want me to do
> about it? Again, this has never been advertised as being compatible with
> systemd -- the worst invention in the history of Linux.

I 100% agree with you evaluation of systemd.  I'm going to be "going
production" with my canboat installation soon and will be spending more
time working on things like this, so you should expect some updates from me.

closed 2019-04-04 by keesverruijt