canboat/canboat #94: PGN 65285 for Lowrance Temperature sensor doesn't support Temperature Source

lfrancois, 2017-09-22:
So far the temperature source is always returning 0, I am going to start to document here the value the sensor is using based on the different values available in B&G Zeus 2 interface.

```
{ "Lowrance: Temperature", 65285, false, 8, 0,
  { { "Manufacturer Code", 11, RES_MANUFACTURER, false, "=140", "Lowrance" }
  , { "Reserved", 2, RES_NOTUSED, false, 0, "" }
  , { "Industry Code", 3, RES_LOOKUP, false, "=4", "Marine Industry" }
  , { "Temperature Instance", 4, 1, false, 0, "" }
  , { "Temperature Source", 4, 1, false, 0, "" }
  , { "Actual Temperature", BYTES(2), RES_TEMPERATURE, false, "K", "" }
  , { 0 }
  }
}
```

lfrancois, 2017-09-22:
Engine Room:
```
1969-12-31-16:36:21,165 0   5 255  65285 Lowrance: Temperature:  Manufacturer Code = Lowrance; Industry Code = Marine Industry; Temperature Instance = 3; Temperature Source = 0; Actual Temperature = 22.30 C (72.1 F)
1969-12-31-16:36:21,165 0 005 255  65285 : 8c 80 03 69 73 ff ff ff 00
```

lfrancois, 2017-09-24:
It looks like `temperature instance` and `temperature source` are inverted.

canboat, 2017-09-24:
Nah, it looks as if there is no Temperature Instance in the Lowrance version, as setting the source to unknown makes 8 bits go 1 (3rd byte goes to 'ff'.)

commit 8955d23f (2017-09-24): Fix #94
closed 2017-09-24 by canboat