canboat/canboat #105: Negative-size-param in parseRawFormatAirmar

gy741, 2018-01-20:
Hello.

I found a Negative-size-param bug in canboat.

Please confirm.

Thanks.

OS: CentOS 7 64bit
Version: 4efdeef5f81e52c5a6d729d524ddbf577ca4a51f
PoC Download: [crash_parseRawFormatAirmar.zip](https://github.com/canboat/canboat/files/1648695/crash_parseRawFormatAirmar.zip)

Steps to reproduce:
1.Download the .POC files.
2.Execute the following command
   :  analyzer <  $POC

```
=================================================================
==44945==ERROR: AddressSanitizer: negative-size-param: (size=-1)
    #0 0x4c81ab in __asan_memcpy /home/karas/src/llvm/projects/compiler-rt/lib/asan/asan_interceptors.cc:453
    #1 0x5174e3 in parseRawFormatAirmar /home/karas/canboat/analyzer/pgn.c:450:3
    #2 0x519bfa in main /home/karas/canboat/analyzer/analyzer.c:334:10
    #3 0x7fb1d4d9cc04 in __libc_start_main (/lib64/libc.so.6+0x21c04)
    #4 0x41af9b in _start (/home/karas/canboat/rel/linux-x86_64/analyzer+0x41af9b)

Address 0x7fffe697cc20 is located in stack of thread T0 at offset 32 in frame
    #0 0x51853f in main /home/karas/canboat/analyzer/analyzer.c:101

  This frame has 3 object(s):
    [32, 2032) 'msg' <== Memory access at offset 32 is inside this variable
    [2160, 2416) 'm'
    [2480, 2484) 'len'
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: negative-size-param /home/karas/src/llvm/projects/compiler-rt/lib/asan/asan_interceptors.cc:453 in __asan_memcpy
==44945==ABORTING
```

=================
[Acknowledgement]
This work was supported by ICT R&D program of MSIP/IITP. [R7518-16-1001, Innovation hub for high Performance Computing]

canboat, 2018-01-21:
Confirmed.

keesverruijt, 2019-02-02:
This was fixed some time ago.

closed 2019-02-02 by keesverruijt