canboat/canboat #111: Problem with "analyzer" when running cross compiled to OpenWRT MIPS (big endian)

captcha1, 2018-02-11:
I cross compiled canboat to OpenWRT, and analyzer returns incorrect results : 

echo 2016-02-28T19:57:02.480Z,2,129025,33,255,8,eb,85,63,08,c8,b5,ab,db | ./analyzer

2016-02-28T19:57:02.480Z 2  33 255 129025 Position, Rapid Update:  Latitude = -147566441688.3081360; Longitude = -429.4967296

The correct result is : 

2016-02-28T19:57:02.480Z 2  33 255 129025 Position, Rapid Update:  Latitude = 14.0740075; Longitude = -60.9503800

I suspect an "endian" issue.

BTW, cross compiling is easy, just set $CC before running make : 

export CC=${CROSS_COMPILE}gcc

... I've attached a script to that downloads the OpenWRT cross compilers and sets $CROSS_COMPILE

captcha1, 2018-02-11:
Oops ... attached the script

[owrt-mips_24kc.txt](https://github.com/canboat/canboat/files/1714717/owrt-mips_24kc.txt)

keesverruijt, 2018-07-24:
Should be fixed now.

commit e59dea30 (2018-07-24): Fix #111

Polish the NMEA0183 output code.
closed 2018-07-24 by keesverruijt