canboat/canboat #113: Add support for PGN 127252 (Heave)

jonross09, 2018-04-06:
I found this snippit of text 

> 127252 Heave
> Field # Field Description
> Vertical displacement perpendicular to (smooth, wave-free water on) the earth’s surface.
> 1 Sequence ID
> 2 Heave
> 3 Delay
> 4 Delay Source
> 5 NMEA Reserved

From this document

https://www.nmea.org/Assets/20151026%20nmea%202000%20pgn_website_description_list.pdf

I'm not exactly sure of the format of the .h file, and it seems like the XML & JSON are generated from the .h file.

Here are some examples 

```
2017-07-15-17:22:28.039,3,127252,11,255,8,00,f8,ff,ff,ff,ff,7f,fd
2017-07-15-17:22:28.178,3,127252,11,255,8,00,fa,ff,ff,ff,ff,7f,fd
2017-07-15-17:22:28.277,3,127252,11,255,8,00,fc,ff,ff,ff,ff,02,ff
2017-07-15-17:22:28.376,3,127252,11,255,8,00,fe,ff,ff,ff,ff,02,ff
2017-07-15-17:22:28.426,3,127252,11,255,8,00,00,00,ff,ff,ff,02,ff
```

and from another day

```
`2017-08-09-00:04:38.319,3,127252,11,255,8,00,08,00,ff,ff,ff,7f,fd
2017-08-09-00:04:38.418,3,127252,11,255,8,00,0e,00,ff,ff,ff,7f,fd
2017-08-09-00:04:38.567,3,127252,11,255,8,00,13,00,ff,ff,ff,7f,fd
2017-08-09-00:04:38.617,3,127252,11,255,8,00,18,00,ff,ff,ff,01,ff
2017-08-09-00:04:38.764,3,127252,11,255,8,00,1d,00,ff,ff,ff,7f,fd
`
```
Let me know how else I can help.

keesverruijt, 2018-10-21:
I have recently obtained a device that produces heave, so this will be added soon. The examples given above did not enable me to determine the precise scale of the numeric value for heave.

neidenmark, 2019-07-02:
Here are a few more Heave examples from my boat (using YNDU RAW)
`10:50:07.240 R 0DF1140A 00 11 00 FF FF FF FE FF
10:50:07.339 R 0DF1140A 00 11 00 FF FF FF 7F FD
10:50:07.442 R 0DF1140A 00 11 00 FF FF FF FE FF
10:50:07.540 R 0DF1140A 00 11 00 FF FF FF 7F FD
10:50:07.690 R 0DF1140A 00 11 00 FF FF FF 7F FD
10:50:07.739 R 0DF1140A 00 11 00 FF FF FF 7F FD
10:50:07.801 R 0DF1140A 00 11 00 FF FF FF FE FF
10:50:09.708 R 0DF1140A 00 0F 00 FF FF FF FF FF
10:50:09.101 R 0DF1140A 00 0F 00 FF FF FF 7F FD`

keesverruijt, 2021-01-27:
Fixed in issue #222

closed 2021-01-27 by keesverruijt