canboat/canboat #115: Yacht Devices USB N2K Gateway

thorsteinssonh, 2018-04-15:
I'm wondering if canboat can be already be used with a nmea USB CDC-ACM type device.
Because they are ACM they have advantage of not requiring any driver.
Yacht Devices provide one such type of NMEA USB,

http://www.yachtd.com/products/usb_gateway.html

sbender9, 2018-04-15:
Should be possible. Can you is an example of what they output?

thorsteinssonh, 2018-04-15:
I'm not too clear yet but receiving this device for testing next week.
The manual gives quite a few uptions http://www.yachtd.com/products/usb_gateway.html
But one mode is RAW, I suspect (hope) it's just contiguous binary messages.

keesverruijt, 2018-04-18:
We'd welcome you contributing the equivalent of `actisense-serial`. If you start with the source code of that program it shouldn't be too hard.

thorsteinssonh, 2018-04-19:
Thanks, I definitely will contribute if end up applying the canboat source. Thanks for the advice. Closing the issue.

closed 2018-04-19 by thorsteinssonh

sbender9, 2018-04-19:
I loooked at the docs today. Should be pretty easy to do. What are you using canboat for/with?

sbender9, 2018-04-19:
I sent an email to asking them to send me a test unit. They would be stupid not to send me one. Would love to add another option for SignalK node server...

thorsteinssonh, 2018-04-19:
Hi Scott,

I'll possibly be using to decode and read N2K weather instrument and
GPS/Nav data.
However, possibly the standard is simple enough just to write support for
these directly into my applications.
I prefer to deal with network traffic like this using Go-lang, possibly
also Python for POC stage.

Hope they give you a trial version.
I noticed this N2K-USB thing gets sent from Russia, and is crazy slow.
Already gone through 10 sorting offices, and 2 customs offices,
and still hasn't left Russia. :)

Yours,
Hrob.

timmathews, 2018-04-19:
@thorsteinssonh I've been meaning to extract the n2k parsing part of https://github.com/timmathews/argo into its own library for some time. It follows the original canboat approach of having all of the PGN definitions compiled in, but converting it to the canboatjs approach of reading in a JSON definition would not be difficult (however, for embedded devices having the data compiled in is usually simpler).

Ideally, I'd like to see a set of N2K parser libraries with essentially the same API for all the most popular languages (or maybe just one library with bindings for other languages).

Let me know if you'd be interested in working on that.

thorsteinssonh, 2018-04-20:
Hi Tim,

Yes, it looks like there is some need for a "'lowest common denominator'
N2K library -- just a plain parser and encoder.
I see these options:

1. writing methods for a few languages in parallel.
2. writing it in C or Go, but making it very easy to write your own bindings
3. using a portable parsing language like  ( kaitai.io  ,... any others? )

I'd be very happy to participate -- at first may require a bit of
discussion though.
It's a little hard for me to tell yet what kind of implementation would
best, until I start experimenting.
Hopefully I can start working with some live data in a week or two.

Yours,
Hrob.

tkurki, 2018-04-20:
Hrob,

Did you look into canboatjs? The cool part about it is that it is 100% driven by the pgn structures in canboat (without js...): so when somebody updates or adds a pgn definition in canboat propagating the change to canboatjs is very easy and no code changes are necessary. 

I believe one master pgn database would be the overall optimal solution. Implementing logic similar to canboatjs in Go is not a huge amount of work, or what I hear about productivity in Go is totally false. Or if you feel you need something compiled create a code generation toolchain to achieve that end.

The same pgn database can be used to create a generic encoder, as implemented in canboatjs.

>making it very easy to write your own bindings

Why would you do that and not propagate the additions upstream? The only reason I can think of is some extension only for your private use, but a solution like one outlined above would support that almost by nature. Afaik there hasn't been much demand for that, but then again the private extension would not be public. Adding support for new pgns is pretty straightforward already.

thorsteinssonh, 2018-04-20:
Hi,
I wasn't advocating any one approach or language.  Canboat and canboatjs
may easily accommodate our eventual dev.
What you are suggesting regarding a database of message definitions that
can be applied to any code or platform sounds excellent.
I have not yet dived into canboatjs but will make sure to test it out.

Cheers,
Hrob.

thorsteinssonh, 2018-04-20:
Hm, I was a bit confused by your question for a bit,
but now I think I see what you are asking, by "my own binding" I certainly
did not mean closed source :)
I would hope that everyone share their cross language binding
implementations.
What I meant to convey was that simplicity of binding often depends on
language and choice of syntax.
Often it helps that a library be exposed in a certain way to make it more
cross compatiple.
One case in point is objects ands classes -- they are normally not as
exportable as basic methods / functions.