canboat/canboat #116: PGN 127258 appears broken

zubenubi, 2018-04-18:
I have versions of canboat. The earlier correctly interprets PGN 127258:

2018-04-18-15:32:29.661 7  35 255 127258 Magnetic Variation:  SID = 140; Source = WMM 2005; Age of service = 2018.04.18; Variation = -14.9 deg

The other is not correctly interpreting the same PGN:

2018-04-18-15:32:38.175 7   1 255 127258 Magnetic Variation:  SID = 178; Source = WMM 2005; Age of service = 2018.04.18; Variation = -1702909974669.2 deg

Both of these output strings are from analyzer via candump2analyzer. 

Im not sure how to determine the version of each install of canboat in order to try to figure out where this might have been introduced.

keesverruijt, 2018-04-18:
Latest version works for me on macOS:

    2016-04-09T16:41:38.014Z 7 127 255 127258 Magnetic Variation:  SID = 0; Source = WMM 2010; Variation = 1.0 deg
    2016-04-09T16:41:38.065Z 7  18 255 127258 Magnetic Variation:  SID = 121; Source = Unknown; Variation = 1.1 deg
    2016-04-09T16:41:38.067Z 7  21 255 127258 Magnetic Variation:  SID = 112; Source = Unknown; Variation = 1.1 deg

Please compile from latest source and try again.

keesverruijt, 2018-10-21:
This is another instance where versioning the product (as requested in issue #123) would have made it easier to see what went on.

keesverruijt, 2019-02-02:
Hi @zubenubi I think this is the same issue as #148. Please re-open if you still have issues when using v1.2.0.

closed 2019-02-02 by keesverruijt