canboat/canboat #117: Yocto layer for canboat

shantanoo-desai, 2018-05-04:
Hello,
I am curious as to if it makes sense to perhaps look into a __Yocto__ layer for canboat.

This just might make canboat more _portable_ to other hardware platforms. I wish to know if there are already efforts from the community in this direction? If not, are there any yocto developers in the mix to help ?

### Findings
- I have found a `meta-signalk` experimental layer on [github](https://github.com/overtane/meta-signalk)

shantanoo-desai, 2018-05-04:
Irrespective of whether or not a layer is available I have created an experimental one [meta-canboat](https://github.com/shantanoo-desai/meta-canboat).
And a [StackExchange Query](https://stackoverflow.com/questions/50177447/cannot-execute-binary-file-exec-format-error-for-a-custom-yocto-layer) for the Initial Error I got whilst building it which might need some resolution if someone is interested

keesverruijt, 2018-05-23:
I hadn't heard of Yocto, but it sounds interesting. If you fix the cross-compiling issues I'm happy to receive pull requests.

shantanoo-desai, 2018-05-24:
@keesverruijt have actually found a solution for the cross-compiling error. I was thinking of keeping it the code more towards a stand-alone repository. 

In short with Yocto you can make a Linux OS for the Raspberry Pi where you specify only necessary stuff like canboat binaries, node, npm, python etc.

shantanoo-desai, 2018-05-24:
@keesverruijt  [meta-canboat](https://github.com/shantanoo-desai/meta-canboat) is the repository which has a yocto layer for all the canboat binaries.

I will do some tests on a [PHYTEC Mira Board](http://www.phytec.eu/product/single-board-computer/phyboard-mira/) and on a raspberry pi too.

If these tests turn out successful, enthusiasts can now use the layer to create their custom embedded-linux OS for different types of single board computers (not just raspberry pi); (and perhaps) making CANBOAT available for [customized hardware](https://www.yoctoproject.org/docs/2.5/brief-yoctoprojectqs/brief-yoctoprojectqs.html#customizing-your-build-for-specific-hardware)

closed 2018-06-03 by shantanoo-desai