canboat/canboat #118: SOG reported by N2KD for NMEA0183 stream is wrong.

mmelsas, 2018-05-16:
In n2kd application the NMEA0183 stream is geeting  wrong value for SOG.  It's probably because PGN 129026 uses SOG number in unit of m/s and this number is also put without any conversion to analyzer output in the JSON format. However it seems that the other conversion procedure "nmea0183VTG" in gps_ais.c  treats that number as SOG in knots and will put it directly to 0183 message as knots and also applies conversion KNOTS_TO_KMH to convert this number to km/h  applies conversion . I was testing it with Garmn plotter and Actisense NGT-1 connected to my laptop.  

BR,
Margus

keesverruijt, 2018-05-23:
Hi Margus, good point. A number of contributed NMEA0183 conversions have unit issues, not just the SOG.

keesverruijt, 2018-07-24:
This should be fixed now.

keesverruijt, 2018-07-24:
See commit e59dea3053cc1134207141b7cbe86d76d35e60be

closed 2018-07-24 by keesverruijt