canboat/canboat #121: Can one log more than one PGNs with the analyzer flag

shantanoo-desai, 2018-09-03:
[Wiki for `analyzer`](https://github.com/canboat/canboat/wiki/analyzer) suggests that using PGN value as a parameter to analyzer will log onlyt that particular output.

It works for example

    actisense-serial -r /dev/ttyUSB0 | analyzer -json 127245

will log only Rudder values for me.

but does something:

     actisense-serial -r /dev/ttyUSB0 | analyzer -json PGN1 PGN2

work to log only the information for `PGN1` and `PGN2`?

keesverruijt, 2018-09-03:
Sorry, this is not supported. The single PGN argument is meant for use during development, not so much in general use.

You can stream the data to n2kd then interrogate this and use a JSON query tool to select the PGNs that you want, or use a JSON query tool on the output of analyzer directly.

tkurki, 2018-09-03:
Try 'actisense-serial -r /dev/ttyUSB0 | grep 'PGN1\|PGN2' | analyzer -json'.

shantanoo-desai, 2018-09-03:
I mean this works great of course with command line utilities like `grep` and `jq` but was curious. In a way I am creating a python script which reads the actisense stuff and compares the json dump with only user defined PGNs and then saves it to InfluxDB. 

If you guys have any suggestions or certain tweaks that might be useful or good then let me know. We can use this thread for some scripts that might be useful for the CANBOAT community.

shantanoo-desai, 2018-09-03:
wasn't aware of that! will look into it.  thanks.

closed 2018-09-07 by keesverruijt