canboat/canboat #129: AIS encoder/!AIVDM in n2kd

free-x, 2018-10-29:
Hi,

whether this will eventually be implemented
``` 
// This does not work yet. It needs to be encoded to format like !AIVDM,1,1,,B,177KQJ5000G?tO`K>RA1wUbN0TKH,0*5C - from
  // http://catb.org/gpsd/AIVDM.html nmea0183CreateMessage(msg183, src, "PVDM,%d%d%d,%s,%s,%s,%s,%s,%.3f,%s,%s", utc_time->tm_hour,
  // utc_time->tm_min, utc_time->tm_sec, mmsiString, latString, lonString, sogString, headingString, cog, rateOfTurnString,
// navigationStatusString);
```
i have already searched for the AIS-encoder implementation in C. Found this only in gpsd(http://git.savannah.gnu.org/cgit/gpsd.git/tree/pseudoais.c) . The question is whether it can be used without further ado, because gpsd was developed under BSD license but "canboat" is under GPL-3

Regards
free-x

keesverruijt, 2018-10-29:
The AIS messages contain binary data, encoded using a variation of Base64 encoding. I am in the process of adding base64 encoding/decoding, it wouldn't be hard to add the mapping used by AIS. This will hit the master branch within the next 2 weeks.

The gpsd code doesn't fit our use case anyway, but it will help establishing the bit patterns in the AIS ITU format.

svdrummer, 2018-10-29:
Would this be needed in CANboat. To transmit in AIS, all owners need a
licence number to be entered into the AIS tranciever at time of
installation. If you are able to receive AIS, that data is already encoded
and formatted.

keesverruijt, 2018-10-29:
@svdrummer the use case is where you have an AIS connected only to NMEA 2000, not 0183, and you have a program that does not handle Signal K or CANboat style NMEA 2000 data, e.g. it can only process NMEA0183 style `!AIVDM` messages.

free-x, 2019-03-20:
#156 

hopefully it will be merged soon

larsujensen, 2019-03-21:
As a short time solution I suppose it is possible to clone my fork of canboat today, checkout the branch ais and build it. Just do not rely on me keeping my fork updated with canboat. And keep in mind that it is a "prototype", take it as it is.

free-x, 2019-03-22:
Hi @larsujensen,
I want to thank you for your great job. I checked out your "ais" branch and played with known samples. Only candumpSample3.txt contains AIS information.

```
cat ../../samples/candumpSample3.txt  | ./candump2analyzer | ./analyzer -json | ./n2kd -d > test.txt 2>&1
cat test.txt | grep "^!" | sort -u 
```

Here is result
```
!AIVDM,1,1,,A,13flrH0000OrIFbJJ2n=M7g805s0,0*46
!AIVDM,1,1,,A,14SeST001OOqUOdJHHAkqC5d0HPN,0*49
!AIVDM,1,1,,A,33P;n8A000Or7MpJGLhqPD>60DNJ,0*69
!AIVDM,1,1,,A,402;bEAv68gh1wqRdDJFBao0081<,0*0A
!AIVDM,1,1,,B,13flrH0000OrIFbJJ2nMbWgH00Sw,0*68
!AIVDM,1,1,,B,402;bEAv68ggkwqRdDJFBao008NO,0*50
!AIVDM,1,1,,B,402:r`1v68ggowrJ3TJJL?W00HPL,0*45
!AIVDM,2,1,1,B,53Jsir02=tfcTP7?C7@p5HTu>1@5E9E<0000001?H@OF:,0*23
!AIVDM,2,2,1,B,6MU0ND1@QhP000000000000000,2*2A
!AIVDO,1,1,,A,B3HpfmP00OvO=o6U9WsQ3wR5oP06,0*27
!AIVDO,1,1,,A,B3HpfmP00OvO=q6U9WgQ3wqUoP06,0*6E
!AIVDO,1,1,,A,B3HpfmP00?vO=o6U9WoQ3wQ5oP06,0*48
!AIVDO,1,1,,A,B3HpfmP00?vO=oVU9WoQ3wP5oP06,0*29
!AIVDO,1,1,,A,B3HpfmP00?vO=p6U9WkQ3wsUoP06,0*11
!AIVDO,1,1,,A,B3HpfmP00?vO=p6U9WoQ3wtUoP06,0*12
!AIVDO,1,1,,A,B3HpfmP00?vO=pVU9WkQ3wrUoP06,0*70
!AIVDO,1,1,,A,B3HpfmP00?vO=qVU9WgQ3wpUoP06,0*7F
!AIVDO,1,1,,A,B3HpfmP00?vO=r6U9WcQ3woUoP06,0*07
```

Some of Online Decoder have problem to decode. The best was at https://www.maritec.co.za/tools/aisvdmvdodecoding/

Could you provide more samples?

Regards
free-x

larsujensen, 2019-03-24:
Yes, online decoders often does not provide desired results. I believe it is the decoders, not then sentences generated. I have used the maritec decoder you mention, and there is a problem also in that decoder. But ultimately the "sharp" decoders are the decoders in the software we use, and in our case it is opencpn. So far I have not seen any problem i opencpn.

I can see that the sample you have is the sample I used in the debugger.  Although it is rather few sentences it actually runs most of the code. Anyhow, I for a long time have planned to make my own candump to get some more data. So I did this yesterday. It is a few minutes of can bus traffic, but the result can be quite overwhelming, I think it contains more than 1500 unique ais messages. I hope my drag and drop include works.

The many duplicates produced by the debug option in n2kd is a problem. If you remove them with sort -u you will unfortunately spoil the ability to decode multi-sentence ais messages. I think that a message with n sentences should be picked from the file the n-th time the first sentence i the message appears. n is 1 and 2 in the file.

Regards
/Lars

[candump-2019-03-23_140244.log](https://github.com/canboat/canboat/files/3000205/candump-2019-03-23_140244.log)

closed 2021-01-28 by keesverruijt