canboat/canboat #143: Battery Configuration Status has several errors

ksltd, 2018-12-10:
There are numerous errors here. I've gleaned some of the correct information from [this](https://www.maretron.com/support/manuals/DCM100UM_1.5.html) document.
Here are some of the more obvious problems:
- The length is wrong but I'm unsure of the correct length but it looks to be around 10; it's certainly a Fast Packet
- Battery Type is an enumeration: Flooded, Gel, AGM
- Supports Equalization seems to be only 1 bit in length, a boolean, and so the reserved field is likely 7 bits
- Nominal Voltage is an enumeration: 6V, 12V, 24V, 32V, 36V, 42V, 48V.
- Chemistry is an enumeration: Flooded, Li, NiCad, ZnO, NiMH
- Temperature coefficient is integer %/ºC and likely only 8 bits
- [This](https://github.com/ttlappalainen/NMEA2000/issues/46) suggests Peukert's Exponent is encoded as one byte
- Charge efficiency is integer percent and likely only 8 bits

ksltd, 2018-12-10:
> Can you add ‘proof’ in the form of raw logs, preferably as pull requests, and with pgn 126996 so that it is clear which device sends the pgn you think is handled incorrectly. This prevents others from claiming that your suggestions are incorrect.—

No. The problem described can be verified by inspection - the length of the datagram is described as 8, but there is more than 64 bits of data described by the field attributes. Wayne Meretsky
Kinetic Scientific
www.kineticscientific.com
Skype:  waynemeretskyUSA: +1   415 218 8103

ksltd, 2019-10-02:
I can confirm that:
- Battery Configuration Status (127513) is a Fast Packet with length of 10 bytes
- Nominal Voltage field is 8 bits and is an enumeration, not a value
- Peukert's Exponent field is 8 bits
- Charge efficiency field is 8 bits

closed 2020-02-28 by keesverruijt

arks-public, 2022-06-08:
I used N2KAnalyzer v3.2.0.3 to analyze Battery Configuration Status (127513) given different data lengths and data contents.

Here is what I found:

- The order and lengths of the data fields in pgn.h are correct (as of last commit October 2021)
- Seems like this must be a fast packet, even though there are only 8 bytes of data. If I send more than 8 bytes, the extra data is shown to be unused. But if I send the 8 bytes of data as a single frame, the message doesn't show up in N2KAnalyzer.
- The "Supports Equalization" field corresponds with LOOKUP_YES_NO, so it doesn't need a custom lookup table
- For the "Chemistry" field I observed a slightly different lookup table: "0=Lead Acid,1=LiIon,2=NiCad,3=ZnO,4=NiMH"
- The other lookup fields (Battery Type and Nominal Voltage) are correct
- Capacity and Peukert Exponent are unsigned fields
- Temperature Coefficient and Charge Efficiency Factor are signed fields in increments of 1% (so these fields range from -128% to 124% since the 3 missing values having special meanings)

commit 27d6d4f4 (2020-02-28): Fix #143

Add the (AFAIAC) more contentious lookup values that look incomplete
commit af85293c (2020-02-28): Fix #143

Follow implementation given by https://github.com/ttlappalainen/NMEA2000/
commit beab1ed4 (2020-02-28): Fix #143

Document doubts generated by https://github.com/ttlappalainen/NMEA2000/