canboat/canboat #146: Trim level

fencer, 2019-01-09:
Trim level of the outboard engine seems not supported, I have a raw data dump if that can be of any help.

fencer, 2019-01-09:
[signalk-rawdata.log.2019-01-09T00.txt](https://github.com/canboat/canboat/files/2742359/signalk-rawdata.log.2019-01-09T00.txt)
Before I head back to the boat, will this log do? Is it ok if I get the value shown on the MFD for the last reading in the log?

fencer, 2019-01-09:
[8M0071193_SC_Gateway-0113_LR (1).pdf](https://github.com/canboat/canboat/files/2742927/8M0071193_SC_Gateway-0113_LR.1.pdf)
found this document on the Mercury site, the PGN is 127488

sbender9, 2019-01-10:
Ok, so canboat supports this, but node server does not. Sorry to make you do this again, but can you open an issue at https://github.com/SignalK/n2k-signalk

I can add trim to https://github.com/SignalK/n2k-signalk/blob/master/pgns/127488.js unless you want to give it a try?

fencer, 2019-01-10:
looks quite straight forward, I'll submit a pull request. 😀

closed 2019-02-02 by keesverruijt