canboat/canboat #147: makefiles?

jcdocka, 2019-01-17:
I am trying to use this program for a senior design project. I have downloaded all of the files as a zip and extracted them all. When I try to compile though, all of the makefiles are missing key cflags and the some c files are missing key includes that need to be linked to .h files. Am I doing something incorrectly and/or do I need to compile the program in a certain way?

keesverruijt, 2019-01-17:
Too little information, I'm afraid. The answer is Yes and Yes.

Have you read the installation instructions https://github.com/canboat/canboat/wiki/Building

keesverruijt, 2019-01-17:
Your question is phrased this way: My car doesn't start. What is wrong?

You need to:
- provide details of your environment and setup
- show the exact command that you are trying and its output.

If you don't do this in the open source world, don't be surprised that you are ignored.