canboat/canboat #153: LOOKUP_GNS_INTEGRITY supposedly wrong

nohal, 2019-02-22:
According to http://www.cruisersforum.com/forums/f134/feature-requests-30931-185.html#post2831847 https://github.com/canboat/canboat/blob/421330cae4df7431fa1b9d62f033845a5e7b009a/analyzer/pgn.h#L230 should be

```
#define LOOKUP_GNS_INTEGRITY (",0=Safe,1=No integrity checking,2=Caution")
```

webmasterkai, 2019-02-22:
https://hemispheregnss.com/gnssreference/GNSSPositionData.htm has the following:

0x0 No integrity checking
0x1 Safe
0x2 Caution
0X3 Unsafe

webmasterkai, 2019-02-22:
[Maretron manual](https://www.maretron.com/support/manuals/GPS200UM_1.3.html) has this to say about the integrity field on NMEA 2000 PGN 129029 – GNSS Position Data:
>  9:    Integrity – This field always contains a value of 0 (no integrity checking).

keesverruijt, 2019-02-22:
@webmasterkai  that is a great source. Adapting canboat this weekend accordingly!

Tehani1, 2019-02-22:
Data field (from 0, byte 30, bits 0 and 1) in PGN 129029 sent from a Raymarine A-series multifunction screen.
The captures were made with a logical analyzer.

This information has also been verified in the files of the CanBoat project: candumpSample1.txt and candumpSample2.txt

Tehani1, 2019-02-23:
I apologize, I had an error in the byte count. "GNSS integrity" starts at message bit 256, this is byte 32.
Looking in different sources I have observed the coincidence with the documentation of Maretron, and of course, it says the same in pgn.h
It seems that this field does not have a useful translation to the sentences: GLL, RMA, RMC and the derivatives for the calculations in navigation: APB, RMB, XTE.
My concern has been to find an equivalence for the field "Data Valid" of these 6 sentences that I mentioned, which is the only one existing for versions of NMEA0183 prior to specification 3.0, and I thought it logical to think that a 0 in the GNSS integrity field corresponded to that "Data Valid" because I detected it together with "Position Fix" in GNS Method. But it is also possible that GNSS integrity is always at 0.
GNS sentence on NMEA0183 has the same "GNSSMethod" information as PGN 129029, so there's no problem.
I think that the best way to interpret "Data Valid" and avoid problems is to ignore GNSS integrity, and use the own indications of NMEA0183:
A = Data valid, V = Loran-C Blink or SNR warning
V = General warning flag for other navigation systems when a reliable fix is ​​not available
And:
The positioning system Mode Indicator field supplements the positioning system Status fields, the Status fields shall be set to V = Invalid for all values ​​of Indicator mode except for A = Autonomous and
D = Differential. The positioning system Mode Indicator and Status fields shall not be null fields.

So this equivalence seems to be coherent:
129029 (GNSSMethod)      Mode Indicator (N0183> = 3.0)     Data valid
0x0 No GPS                          N = Data not Valid                          V
0x1 GNSS fix                         A = Autonomous mode                 A
0x2 DGNSS fix                       D = Differential mode                    A
0x3 Precise GNSS                  D = Differential mode                    A
0x4 RTK fixed integer            D                                                     A
0x5 RTK float                         D                                                     A
0x6 Estimated (DR) mode     E                                                      V
0x7 Manual input                  M                                                    V
0x8 Simulate mode                S                                                     V
0xE Error                                N                                                     V

sbender9, 2019-02-23:
So no changes to canboat, right?

Want to close this?

Tehani1, 2019-02-23:
I think so.
Again, I apologize for any doubts in the CanBoat documentation.
I think it is better to use GNSS Method exclusively to have a data integrity information.
Thank you.

webmasterkai, 2019-02-23:
@Tehani1 The following graphic was helpful to me in using 129029 to create GGA.
[image]

@sbender9 It's possible there is another value `0X3 Unsafe` that could be added to `LOOKUP_GNS_INTEGRITY` but I haven't confirmed yet. Who has a copy of the NMEA PGN file? I probably should just buy a copy.

Tehani1, 2019-02-23:
In N2K, when a 2-bit field has a value of 0x3 it means that it does not contain data.
"Unsafe" is an information. It would be a data.

Tehani1, 2019-02-23:
Contents in 129029 "Method GNS" are identical to GPS Quality Indicator field.

129029, 2nd source:
https://hemispheregnss.com/gnssreference/GNSSPositionData.htm

Tehani1, 2019-02-23:
I do not want to bother you anymore.
My goal has been to complete RMC and GLL to the maximum. I needed to supply "Status" information.

closed 2020-02-28 by keesverruijt