canboat/canboat #161: Missing AIS Transceiver information in PGN 129809 and PGN 129810

larsujensen, 2019-04-13:
PGNs 129809 and 129810 are both marked as incomplete. These are very common PGNs, providing ais class B static data reports. One of the missing fields is "AIS Transceiver information", which is used in encoding nmea 0183 ais sentences, and may possibly be important in some plotters, as it tells whether the information is from own or foreign vessel.

In PGN 129809 the fields are 1 - Message ID, 2 - Repeat Indicator, 3 - User ID, 4 - Name, 5 - AIS Transceiver Information, 6 - NMEA Reserved, 7 - Sequence ID. As the first four are decoded in canboat, and the one needed is the fifth field, which should be as in other AIS pgns, it should be easy to add this field. The last two fields are not required for generating nmea 0183 ais sentences.

In PGN 129810 AIS Transceiver Information is field number 14, but again canboat decodes all fields up to and including field 13, so it should be easy to add this field. And again, the last two fields are NMEA Reserved and Sequence ID, not used when generating nmea 0183 ais sentences.

sbender9, 2019-04-13:
Do you have details on these fields? Size, type, lookup, etc?

larsujensen, 2019-04-14:
Concerning AIS Transceiver information this field is, I think, in every AIS PGN, and it would be very surprising if it was coded differently in PGNs 129809 and 129810. Copied from pgns.json it looks like this in PGN 129038:
          {
            "Order":12,
            "Id":"aisTransceiverInformation",
            "Name":"AIS Transceiver information",
            "BitLength":5,
            "BitOffset":163,
            "BitStart":3,
            "Type":"Lookup table",
            "Signed":false,
            "EnumValues":[
              {"name":"Channel A VDL reception","value":"0"},
              {"name":"Channel B VDL reception","value":"1"},
              {"name":"Channel A VDL transmission","value":"2"},
              {"name":"Channel B VDL transmission","value":"3"},
              {"name":"Own information not broadcast","value":"4"},
              {"name":"Reserved","value":"5"}]},
I have run samples through analyzer -raw, and picked out this:
2019-03-23-13:04:42.228 6 043 255 129809 : a0 1b 18 ea 5b d7 0f 45
2019-03-23-13:04:42.229 6 043 255 129809 : a1 4c 4c 4f 49 53 45 40
2019-03-23-13:04:42.239 6 043 255 129809 : a2 40 40 40 40 40 40 40
2019-03-23-13:04:42.249 6 043 255 129809 : a3 40 40 40 40 40 **03** ff
and marked in bold the byte I think should contain the 5 bits in bits 0-4. The value is thus 3 and this corresponds to a transmission, which I know it should do for this particular selected pgn, i.e. value 2 or 3.
Another sample:
2019-03-23-13:06:06.762 6 043 255 129809 : e0 1b 18 fc 7b d7 0f 57
2019-03-23-13:06:06.770 6 043 255 129809 : e1 41 4b 45 20 4d 45 20
2019-03-23-13:06:06.780 6 043 255 129809 : e2 55 50 40 40 40 40 40
2019-03-23-13:06:06.789 6 043 255 129809 : e3 40 40 40 40 40 **00** ff
Here the corresponding bits are 0 and the value should be 0 or 1 i.e. reception.

There are many PGNs with Sequence ID, fetched from pgns.json PGN 127250:
          {
            "Order":1,
            "Id":"sid",
            "Name":"SID",
            "BitLength":8,
            "BitOffset":0,
            "BitStart":0,
            "Signed":false},
It seems reasonable that Sequence ID should be like this also in PGN 129809.
The second byte in the first frame is 1b which tells us that there are 27 bytes of data in the packet.
Since Sequence ID is supposed to be the last field in PGN129809, and data in can frames end on byte boundaries sequence id should be the **ff** in above samples. Maybe ff signifies no sequence id available?  If so the NMEA reserved field should be bits 5-7 in the bold bytes.

The sample below is for 129810, and I have marked the byte i believe contains AIS Transceiver information, once again i bits 0-4. The value should be 2 or 3 for this selected pgn.
2019-03-23-13:04:52.588 6 043 255 129810 : 00 23 18 ea 5b d7 0f 24
2019-03-23-13:04:52.597 6 043 255 129810 : 01 53 4d 54 44 2a 33 44
2019-03-23-13:04:52.607 6 043 255 129810 : 02 53 46 44 32 34 33 36
2019-03-23-13:04:52.617 6 043 255 129810 : 03 82 00 28 00 28 00 50
2019-03-23-13:04:52.627 6 043 255 129810 : 04 00 00 00 00 00 00 **03**
2019-03-23-13:04:52.637 6 043 255 129810 : 05 ff ff ff ff ff ff ff
In this case there should be 35 bytes of data in the packet. Thus Sequence id should once again be ff, the first byte in frame 5. And NMEA reserved once again should be bits 5-7 in the bold byte.

sbender9, 2019-04-14:
Thanks. I went through the same exercise. Fields added in #163

closed 2019-04-14 by keesverruijt

larsujensen, 2019-04-14:
And an observation about 129038. There are 20 fields in this PGN, the last one being Sequence ID. This is not present in pgns.json, which in total have 216 bits, corresponding to 27 bytes. Once again a sample from analyzer -raw:
2019-03-23-13:04:52.858 4 043 255 129038 : a0 1c 01 d8 f5 d6 0f ff
2019-03-23-13:04:52.867 4 043 255 129038 : a1 ff ff 7f ff ff ff 7f
2019-03-23-13:04:52.877 4 043 255 129038 : a2 fc ff ff ff ff 4d c0
2019-03-23-13:04:52.887 4 043 255 129038 : a3 02 ff ff ff 7f 08 00
2019-03-23-13:04:52.897 4 043 255 129038 : a4 ff ff ff ff ff ff ff
As we can see the length is 1c, i.e. 28 bytes. The 28th byte is the first ff in frame 4.
Now I do not think this is important.

But there are more to it. Up to and including field 15, Navigational Status, pgns.json strictly follows NMEA 2000 Standard for Serial-Data Networking of Marine Electronic Devices. After this pgns.json specifies 3 more fields, 16 - reserved (4 bits), 17 -Regional Application (1 bit), and 18 - Reserved (7 bits).
In the NMEA 2000 Standard the fields are 16 - Special Maneuver Indicator, 17 - NMEA Reserved, 18 - AIS Spare, and 19 - NMEA Reserved. And then finally above mentioned 20 - Sequence ID.

Clearly there are 12 bits that seems to be quite shaky.
In an ais message the special maneuvre indicator is 2 bits coded as:
0 = not available = default, 1 = not engaged in special maneuver, 2 = engaged in special maneuver (i.e.: regional passing arrangement on Inland Waterway)
whatever the interpretation of this is.

I think the definition of the pgn in pgns.json is old, there where earlier a regional reserved field of 3 bits followed by a 2 bit spare in the common block ais message, but this is from revision 3 of ITU-R M.1371 replaced by a 2 bit maneuvre indicator followed by 3 spare bits.
My guess is that the fields in the pgn should be defined as 16 - Special Maneuver Indicator (2 bits), 17 - NMEA Reserved (2 bits), 18 - AIS Spare (3 bits), 19 - NMEA Reserved (5 bits), and 20 - Sequence ID (8 bits). All these bits (except sequence id) are 0 in above sample, which they should be, the default value of the special maneuver indicator is 0, AIS Spare should be 0 according to standard, and NMEA reserved are reasonably set to 0.

PR #163 (2019-04-14): fix: add missing "AIS Transceiver Information" fields to class b static pgns
Fixes #161