canboat/canboat #164: Missing field(s) in PGN 129038

larsujensen, 2019-04-14:
An observation about 129038. There are 20 fields in this PGN, the last one being Sequence ID. This is not present in pgns.json, which in total have 216 bits, corresponding to 27 bytes. Once again a sample from analyzer -raw:
2019-03-23-13:04:52.858 4 043 255 129038 : a0 1c 01 d8 f5 d6 0f ff
2019-03-23-13:04:52.867 4 043 255 129038 : a1 ff ff 7f ff ff ff 7f
2019-03-23-13:04:52.877 4 043 255 129038 : a2 fc ff ff ff ff 4d c0
2019-03-23-13:04:52.887 4 043 255 129038 : a3 02 ff ff ff 7f 08 00
2019-03-23-13:04:52.897 4 043 255 129038 : a4 ff ff ff ff ff ff ff
As we can see the length is 1c, i.e. 28 bytes. The 28th byte is the first ff in frame 4.
Now I do not think this is important.

But there are more to it. Up to and including field 15, Navigational Status, pgns.json strictly follows NMEA 2000 Standard for Serial-Data Networking of Marine Electronic Devices. After this pgns.json specifies 3 more fields, 16 - reserved (4 bits), 17 -Regional Application (1 bit), and 18 - Reserved (7 bits).
In the NMEA 2000 Standard the fields are 16 - Special Maneuver Indicator, 17 - NMEA Reserved, 18 - AIS Spare, and 19 - NMEA Reserved. And then finally above mentioned 20 - Sequence ID.

Clearly there are 12 bits that seems to be quite shaky.
In an ais message the special maneuvre indicator is 2 bits coded as:
0 = not available = default
1 = not engaged in special maneuver
2 = engaged in special maneuver
(i.e.: regional passing arrangement on Inland Waterway)
whatever the interpretation of this is.

I think the definition of the pgn in pgns.json is old, there where earlier a regional reserved field of 3 bits followed by a 2 bit spare in the common block ais message, but this is from revision 3 of ITU-R M.1371 replaced by a 2 bit maneuver indicator followed by 3 spare bits.
Provided the newer revision should be followed, my guess (only a guess) is that the fields in the pgn should be defined as 16 - Special Maneuver Indicator (2 bits), 17 - NMEA Reserved (2 bits), 18 - AIS Spare (3 bits), 19 - NMEA Reserved (5 bits), and 20 - Sequence ID (8 bits). All these bits (except sequence id) are 0 in above sample, which they should be, the default value of the special maneuver indicator is 0, AIS Spare should be 0 according to standard, and NMEA reserved are reasonably set to 0.

closed 2019-04-15 by keesverruijt

negrusti, 2025-01-09:
[candump-2022-06-28_084210Part1.log](https://github.com/user-attachments/files/18358583/candump-2022-06-28_084210Part1.log)
I am seeing 27-byte packets in this capture

keesverruijt, 2025-01-09:
> [candump-2022-06-28_084210Part1.log](https://github.com/user-attachments/files/18358583/candump-2022-06-28_084210Part1.log) I am seeing 27-byte packets in this capture

Yup. Older AIS probably, or at least not sending out the relatively late addition of the last "Sequence ID" field. My Cortex AIS does produce 28 bytes.

Leaving issue closed. (You're lucky I saw the email, updating closed issues doesn't usually register.)

negrusti, 2025-01-09:
@keesverruijt but that means that JSON definition can't be reliably used to parse both variations automatically. Can Sequence ID be marked optional somehow? That still leaves the issue of the fixed length specification though...

keesverruijt, 2025-01-09:
Yes it can, your parser shall have to adjust to reality here, any fields at the end not present are ... not present. I would not know which versions of the NMEA standard which fields have been added, this is just reverse engineered.

keesverruijt, 2025-08-22:
Can you guys tell me what AIS you are using and what NMEA version it claims compatibility with (PGN 126996 field "Certification Level"?) 

This will allow us to start adding an extra field "new in NMEA version X.Y" once we get a feel for the probable version this was added.

uschi2000, 2025-08-22:
Will be back at the boat next week and can try to figure out vendor and version etc. I don't see any 126996 in the data, unfortunately.

PR #165 (2019-04-15): fix: refine and finalize AIS pgns
Fixes #164
commit d633bde4 (2019-04-15): fix: refine and finalize AIS pgns

Fixes #164