canboat/canboat #176: NMEA 2000: PGN 127489 (Engine Parameters, Dynamic)

john993, 2019-07-11:
Hi everyone,
Sorry maybe my question is out of topic, but it is have relation with NMEA and CAN boat.

I have question, how to get the **Engine Percent Load** with this data:

Chn  CAN Identifier  Flg  DLC   D0... 1... 2... 3...   4...   5... 6...   D7   Time
0      166855062        X      8     192  26  0  255 255 255 255 249  35.140360

I look at PGN Explorer, the bit offset is 192, but the data I got from CAN is only 8 byte (64 bit).

Thanks in advance.

keesverruijt, 2020-02-28:
Hi John,

The PGN is a Fast Packet PGN, so they all belong together. There are various places where canboat pulls these back together into a single message.

As you can see the first packet has the first byte 00 followed by 1A. This means: this is the first packet (=00) and the total length is 0x1A bytes. The 6 following bytes are the first 6 content bytes. The remaining packets have 7 content bytes each.

The `candump2analyzer` program doesn't bother with any of this, but `analyzer` does. It recognises the format as one that is `MULTIPACKET_SEPARATE` and reformats the entire message before decoding it.

closed 2020-02-28 by keesverruijt

snnszgzl, 2024-02-16:
Hi, also i have same question,
[image]
picture is show PNG 127489 messages. How can i read Engine Load Percent from here ?

keesverruijt, 2024-02-16:
Please don't hijack (closed) issues, this is not a forum. See new issue.