canboat/canboat #183: Problems parsing PGN126208 and PGN126720 w/ Airmar DST200

lanceberc, 2019-10-23:
Error messages I believe originate from an Airmar DST200 depth/speed/temperature sensor:

```
ERROR 2019-10-23T15:06:55.596Z [analyzer] PGN 126720 malformed packet for 204 received; expected 1 but got 59
ERROR 2019-10-23T15:06:55.596Z [analyzer] Pgn 126208 Field Value: cannot derive variable length from PGN 126720 field # 30
ERROR 2019-10-23T15:06:55.597Z [analyzer] PGN 126720 malformed packet for 8 received; expected 1 but got 59
ERROR 2019-10-23T15:06:55.597Z [analyzer] Pgn 126208 Field Value: cannot derive variable length from PGN 126720 field # 30
```

We do see correctly formed data for depth, boatspeed, and sea temperature.

keesverruijt, 2020-02-28:
Could you upload raw data please, this is not something I'm seeing on my DST200.

keesverruijt, 2020-02-28:
Note, this is a special case. PGN 126720 is basically saying "For this other PGN, I want you to set", or "Can you tell me what you have set it to," or "I have set the value to", etc. and then the data that follows are fields of other PGNs. In this case, either the DST200 is reporting data, or more likely, you have some tool trying to program your DST200. Since it is programming an Airmar specific PGN (126720) that we don't know the exact content of, it is very likely that the above occurs.
Until we have fixed definitions for 126720, the above will happen (and you have these PGNs on the bus, of course.)

keesverruijt, 2020-09-30:
A lot of work has been spent in improving the handling of variable fields, can you try again @lanceberc and if it still fails please upload the raw PGNs that you see that are still decoded incorrectly?

lanceberc, 2020-09-30:
With the "current situation" the boats I have access to are either out of the water for long-term repairs or not racing. I hope we're all back on the water soon and will try again, but it may not be until spring. Thank you for looking into this.

mglonnro, 2021-05-06:
Sorry if/when this is the wrong place to post this, but I'm trying my best to _send_ 126208 to a DST800 (with actisense-serial), but can't get it to work. The idea is to get the DST800 to report the voltage it sees using PGN 65410.

Where could I ask this question without making too much trouble? :)

keesverruijt, 2021-07-29:
@mglonnro NOT somewhere where nobody will notice that you have added a totally different question! Create a new issue and mark it as Question.

closed 2022-10-18 by keesverruijt