canboat/canboat #184: PGN 127251: Negative rate of rotation for RC42 is not working.

violapaul, 2019-11-22:
I noticed that my logs only show positive rotation rates and **very small** negative rotations (essentially clipped). 

Note, not a super C hacker so providing a fix will be tough.  But here is what I have.

I started to dig,  working from the canboat logs and the source code, and I implemented my own conversion in Python (see below).  I get agreement on the values for **positive* angles.,  but **not for negative**.

Log files and JSON file are here: [test.zip](https://github.com/canboat/canboat/files/3881721/test.zip)

The output of the Python is here (code at bottom... nothing fancy).

```
2019-10-19T03:38:13.695Z,2,127251,4,255,8,ff,65,c7,01,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.695Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00364316}}
    Values match 0.00364, 0.00364

2019-10-19T03:38:13.727Z,2,127251,4,255,8,ff,38,4b,02,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.727Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00469775}}
    Values match 0.00470, 0.00470

2019-10-19T03:38:13.795Z,2,127251,4,255,8,ff,65,c7,01,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.795Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00364316}}
    Values match 0.00364, 0.00364

2019-10-19T03:38:13.854Z,2,127251,4,255,8,ff,bf,bf,00,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.854Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00153397}}
    Values match 0.00153, 0.00153

2019-10-19T03:38:13.902Z,2,127251,4,255,8,ff,11,d0,ff,ff,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.902Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': -7.47e-06}}
*** Values don't match -0.00001, -0.00038

2019-10-19T03:38:13.919Z,2,127251,4,255,8,ff,3d,4c,ff,ff,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.919Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': -6.09e-06}}
*** Values don't match -0.00001, -0.00144

2019-10-19T03:38:13.991Z,2,127251,4,255,8,ff,11,d0,ff,ff,ff,ff,ff
{'timestamp': '2019-10-19T03:38:13.991Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': -7.47e-06}}
*** Values don't match -0.00001, -0.00038

2019-10-19T03:38:14.022Z,2,127251,4,255,8,ff,6a,c8,fe,ff,ff,ff,ff
{'timestamp': '2019-10-19T03:38:14.022Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': -4.69e-06}}
*** Values don't match -0.00000, -0.00249

2019-10-19T03:38:14.085Z,2,127251,4,255,8,ff,eb,3b,00,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:14.085Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00047934}}
    Values match 0.00048, 0.00048

2019-10-19T03:38:14.171Z,2,127251,4,255,8,ff,eb,3b,00,00,ff,ff,ff
{'timestamp': '2019-10-19T03:38:14.171Z', 'prio': 2, 'src': 4, 'dst': 255, 'pgn': 127251, 'description': 'Rate of Turn', 'fields': {'Rate': 0.00047934}}
    Values match 0.00048, 0.00048
```

```python
test_log = '/Users/viola/canlogs/OldLogs/test.log'
test_json = '/Users/viola/canlogs/OldLogs/test.json'

with open(test_log) as flog:
    log_lines = [l.strip() for l in flog.readlines()]

with open(test_json) as fjson:
    json_lines = [json.loads(l) for l in fjson.readlines()]

m = 1e-6 / 32.0

for l, j in zip(log_lines, json_lines):
    print()
    print(l)
    print(j)
    byte_sequence = l.strip().split(',')[-7:]
    canboat_rate = j['fields']['Rate']
    hex_string = ''.join(byte_sequence[:4])

    bb = bytes.fromhex(hex_string)
    val = struct.unpack('<i', bb)[0]
    rate = val * m
    match = (canboat_rate - rate) < 0.000001
    if match:
        print(f'    Values match {canboat_rate:.5f}, {rate:.5f}')
    else:
        print(f'*** Values don\'t match {canboat_rate:.5f}, {rate:.5f}')
```

nmea4000, 2019-11-23:
I get the following values from my software.
```javascript
[
  { data: '65,C7,01,00', rate: 0.00364315625 },
  { data: '38,4B,02,00', rate: 0.00469775 },
  { data: '65,C7,01,00', rate: 0.00364315625 },
  { data: 'BF,BF,00,00', rate: 0.00153396875 },
  { data: '11,D0,FF,FF', rate: -0.00038346875 },
  { data: '3D,4C,FF,FF', rate: -0.00143809375 },
  { data: '11,D0,FF,FF', rate: -0.00038346875 },
  { data: '6A,C8,FE,FF', rate: -0.0024926875 },
  { data: 'EB,3B,00,00', rate: 0.00047934375 },
  { data: 'EB,3B,00,00', rate: 0.00047934375 },
  { data: 'EB,3B,00,00', rate: 0.00047934375 }
]
```

violapaul, 2019-11-23:
My bad...  thanks.  And sorry for the trouble.

I had a very old version of analyzer in /usr/local/bin that was likely installed with Signalk (?).  I had the latest source but when I went to install it, it would not install in /usr/bin on my Mac (some sort of security thing and I did not follow up on that).  Looks like it was a bug that was fixed (perhaps long ago).

Thanks,
Paul

closed 2019-11-23 by violapaul