canboat/canboat #187: NMEA0183 with date

wellenvogel, 2020-02-03:
I'm trying to use canboat (n2kd) to convert nmea2k to nmea0183 for the use in my nav software (https://github.com/wellenvogel/avnav).
Currently there seems to be no NMEA0183 record that has a date included (would be RMC,ZDA,TRF referring to https://gpsd.gitlab.io/gpsd/NMEA.html).
Without such a record NMEA0183 based instruments would not be able to display date/time
Would there be a chance to include one of those records?
Maybe RMC would be the best choice - although it would potentially invoke some state handling as the data would have to be collected from different pgns.

tkurki, 2020-02-03:
I'm sure Kees, who is involved with Signal K since the very beginning, doesn't mind me saying that consider using SK Node server, it can convert N2K to Signal K and on to NMEA0183.

 You could even start using this data directly in SK format, to make a start towards using SK extensively?

wellenvogel, 2020-02-03:
Yes, sure signalk is an option and I also implemented it in my software using the json api.
But from the architectural pov it would be much simpler for the user and the data flow would be more easy.

white74, 2020-02-03:
GGA has time from gps in the 1st value

wellenvogel, 2020-02-03:
yes but time only, no date

keesverruijt, 2020-02-03:
I agree with Teppo: signal k is the future...

Still, I am more than willing to accept your pull request if you add this.
But n2kd is essentially "deadish" -- I only intend to add and fix stuff for
my own application.

Kees

commit b1c84477 (2021-01-27): Fix #187: Add ZDA sentence
closed 2021-01-27 by keesverruijt