canboat/canboat #198: Build fail with gcc 10.0 and 10.1: multiple definitions

garyemiller, 2020-06-19:
```
canboat # make
mkdir -p rel/linux-x86_64
for dir in actisense-serial analyzer n2kd nmea0183 ip group-function candump2analyzer socketcan-writer ikonvert-serial; do make -C $dir; done
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/actisense-serial'
cc    -o ../rel/linux-x86_64/actisense-serial -I../common actisense-serial.c ../common/common.c 
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccnHnN9h.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccHcIpyh.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:28: ../rel/linux-x86_64/actisense-serial] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/actisense-serial'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/analyzer'
cc    -o ../rel/linux-x86_64/analyzer -I../common pgn.c analyzer.c ../common/common.c 
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccC8ehc6.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccd4xca6.o:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cc4UqaT3.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccd4xca6.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:30: ../rel/linux-x86_64/analyzer] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/analyzer'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/n2kd'
cc    -o ../rel/linux-x86_64/n2kd -I../common main.c gps_ais.c nmea0183.c ../common/common.c -lm
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cchh4vjl.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccSWR6tn.o:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cc3reyum.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccSWR6tn.o:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccmIoqFl.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccSWR6tn.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:30: ../rel/linux-x86_64/n2kd] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/n2kd'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/nmea0183'
cc    -o ../rel/linux-x86_64/nmea0183-serial  -I../common nmeareader.c ../common/common.c
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cci8crjK.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccWrCbLL.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:36: ../rel/linux-x86_64/nmea0183-serial] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/nmea0183'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/ip'
cc    -o ../rel/linux-x86_64/iptee  -I../common iptee.c ../common/common.c
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccO71TlQ.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccitRDlS.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:36: ../rel/linux-x86_64/iptee] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/ip'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/group-function'
cc    -o ../rel/linux-x86_64/command-group-function -I../common command.c ../common/common.c 
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccY2lUPX.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccyjQPU0.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:29: ../rel/linux-x86_64/command-group-function] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/group-function'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/candump2analyzer'
cc    -o ../rel/linux-x86_64/candump2analyzer -I../common candump2analyzer.c ../common/common.c -lm
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cc94WZrt.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccqor8Yt.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:29: ../rel/linux-x86_64/candump2analyzer] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/candump2analyzer'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/socketcan-writer'
cc    -o ../rel/linux-x86_64/socketcan-writer -I../analyzer -I../common socketcan-writer.c ../analyzer/pgn.c ../common/common.c 
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccr9nJZ4.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccDOmyh7.o:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cc7oyrH6.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccDOmyh7.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:34: ../rel/linux-x86_64/socketcan-writer] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/socketcan-writer'
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/ikonvert-serial'
cc    -o ../rel/linux-x86_64/ikonvert-serial -I../common ikonvert-serial.c ../common/common.c ../common/b64.c 
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/cc24OmXJ.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccjAX0fK.o:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccDdxAUI.o:(.bss+0x0): multiple definition of `sbNew'; /tmp/ccjAX0fK.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:28: ../rel/linux-x86_64/ikonvert-serial] Error 1
make[1]: Leaving directory '/u1/src/GPS/gpsd/canboat/ikonvert-serial'
make: *** [Makefile:29: all] Error 2
spidey /usr/local/src/GPS/gpsd/canboat # gcc --version
gcc (Gentoo 10.1.0-r1 p2) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTI
```

garyemiller, 2020-06-19:
externs are not being properly defined.

Work around: -fcommon

````
# CFLAGS="-fcommon" make
for dir in actisense-serial analyzer n2kd nmea0183 ip group-function candump2analyzer socketcan-writer ikonvert-serial; do make -C $dir; done
make[1]: Entering directory '/u1/src/GPS/gpsd/canboat/actisense-serial'
cc  -fcommon  -o ../rel/linux-x86_64/actisense-serial -I../common actisense-serial.c ../common/common.c 
```

But best to fix the includes.

keesverruijt, 2020-08-09:
Fixed in commits a7cb759 and 8983614.

closed 2020-08-09 by keesverruijt