canboat/canboat #199: Low performance and checksum errors on Raspberry Pi

jdt1, 2020-06-22:
When running canboat on my Raspberry Pi (3B+ and 4), I'm experiencing poor performance and checksum errors. When I connect the exact same network (including actisense NGT-1 adapter) to my laptop (running Ubuntu), everything works great. Some things I read online suggested that canboat should run fine on any of the newer Raspberry Pis, htop also shows low cpu usage (3 - 5 %).

The command I'm running is simply `actisense-serial /dev/ttyUSB0 > tmp.txt`.

I am getting various errors.
```
ERROR 2020-06-16T04:58:35.635Z [actisense-serial] Ignoring message with invalid checksum
ERROR 2020-06-16T04:58:37.451Z [actisense-serial] Ignoring message with invalid checksum
ERROR 2020-06-16T04:58:38.908Z [actisense-serial] Ignoring message with invalid checksum
ERROR 2020-06-16T04:58:39.635Z [actisense-serial] DLE followed by unexpected char 26, ignore message
ERROR 2020-06-16T04:58:39.635Z [actisense-serial] Ignore short command len = 0
ERROR 2020-06-16T04:58:41.909Z [actisense-serial] Ignoring message with invalid checksum
ERROR 2020-06-16T04:58:42.437Z [actisense-serial] Ignoring message with invalid checksum
```

When looking directly at the output of `actisense-serial /dev/ttyUSB0` the output freezes every second or so, like the system is busy. On my laptop, this does not happen. There is a larger packet that arrives every second or so, perhaps this packet is too much for the Pi to handle?

This is a part of a typical output seen on my laptop, with the larger packet arriving at time 15:19:17.929Z

```
2020-06-22T15:19:17.783Z,2,127250,128,255,8,00,ca,56,ff,7f,e7,00,fd
2020-06-22T15:19:17.783Z,3,127251,128,255,8,00,70,f3,f2,ff,00,ff,ff
2020-06-22T15:19:17.929Z,2,129026,127,255,8,00,fd,ca,56,00,00,ff,ff
2020-06-22T15:19:17.929Z,2,129025,127,255,8,ad,9f,3a,1f,db,d1,e4,02
2020-06-22T15:19:17.929Z,2,127250,128,255,8,00,c9,56,ff,7f,e7,00,fd
2020-06-22T15:19:17.929Z,3,127251,128,255,8,00,43,89,f2,ff,00,ff,ff
2020-06-22T15:19:17.929Z,3,129029,127,255,43,00,03,48,a8,65,e0,20,c0,08,cd,56,b5,64,45,07,58,11,33,58,4c,7c,ac,00,50,1f,45,00,00,00,00,00,12,fc,00,64,00,96,00,5c,12,00,00,00
2020-06-22T15:19:17.929Z,6,129540,127,255,147,00,ff,0c,08,50,2c,43,75,fc,08,00,00,00,00,f2,0a,2e,17,7f,25,5c,12,00,00,00,00,f2,0e,ae,0f,8a,54,10,0e,00,00,00,00,f2,15,c5,04,0a,3e,48,0d,00,00,00,00,f2,16,8b,18,94,92,d0,07,00,00,00,00,f2,1b,2e,17,38,64,48,0d,00,00,00,00,f2,20,7f,16,38,46,94,11,00,00,00,00,f2,4a,17,04,68,03,60,09,00,00,00,00,f2,4a,8b,09,59,b5,00,00,00,00,00,00,f2,42,f3,2a,c4,40,54,0b,00,00,00,00,f2,52,45,2a,87,db,00,00,00,00,00,00,f2,41,e8,0a,16,13,28,0a,00,00,00,00,f2
2020-06-22T15:19:17.929Z,3,126992,127,255,8,00,f0,03,48,a8,65,e0,20
2020-06-22T15:19:17.929Z,6,129539,127,255,8,00,d3,64,00,78,00,ff,7f
2020-06-22T15:19:17.929Z,7,127258,127,255,8,00,f6,ff,ff,e7,00,ff,ff
2020-06-22T15:19:17.929Z,3,127257,128,255,8,00,ff,7f,19,f5,4d,f8,ff
2020-06-22T15:19:17.963Z,0,262386,0,0,33,01,0e,00,a1,87,03,00,00,00,00,00,02,05,02,00,00,00,00,00,00,00,00,0c,54,02,02,00,00,00,04,00,00,00
2020-06-22T15:19:17.989Z,2,129026,127,255,8,00,fd,c9,56,00,00,ff,ff
2020-06-22T15:19:17.989Z,2,129025,127,255,8,ad,9f,3a,1f,db,d1,e4,02
2020-06-22T15:19:17.989Z,2,127250,128,255,8,00,cc,56,ff,7f,e7,00,fd
2020-06-22T15:19:17.989Z,3,127251,128,255,8,00,d5,2f,f3,ff,65,e0,20
2020-06-22T15:19:18.080Z,2,129026,127,255,8,00,fd,cc,56,00,00,ff,ff
2020-06-22T15:19:18.080Z,2,129025,127,255,8,ad,9f,3a,1f,db,d1,e4,02
```

What kind of performance can I expect of canboat? Is the performance better if I'm not using actisense-serial directly but using the daemon instead (seems unlikely but I don't know for sure) ?. 

I'm hoping to get 10 Hz output from my GPS unit over N2k. On my laptop this works fine, should this be possible on a Pi or am I just asking too much?

Thank you,

Jim

keesverruijt, 2020-08-09:
I run the entire canboat stack just fine on a RPi 3b, in production since January, replacing a 500 MHz x86 board.

The longer sentences for PGNs 129029 and 129540 are normal.

Not sure what is going on here, maybe there is something fishy going on with the USB drivers.

I take it you are using the latest Raspbian and not some other OS?

keesverruijt, 2021-01-27:
I'm rereading this and I think the most likely problems is that you have the Actisense set to a different baud rate than the default for `actisense-serial` which is 115200 baud; this _used_ to be the default for the NGT-1 as well but newer Actisense firmware has changed this. To avoid breaking existing installations on upgrade, we are not changing the default.

So try

    actisense-serial -s 230400 /dev/ttyUSB0 > tmp.txt

jdt1, 2021-01-28:
Hi Kees, thanks for following up! I am currently not working with NMEA2k but I will probably pick it up again in a few months. Your explanation seems very plausible.