canboat/canboat #201: Visualising data

coconha2, 2020-07-10:
Hi, I have good experience with CAN bus and I am now very interested in NMEA2000. I have a boat with a B&G network and am learning how it works and would like to potentially build a couple of devices for it. This is great project, congratulations!

I just started using Analyzer and was wondering which tool is used to visualise data such as this plot: https://github.com/canboat/canboat/issues/148

sbender9, 2020-07-10:
You should check out Signal K (www.signalk.org) and the Server: https://github.com/SignalK/signalk-server-node

This uses canboat (canboatjs) under the hood. There you can use use influxdb to store history and then visualize it with Grafana: http://openplotter.wikidot.com/grafana

sbender9, 2020-07-10:
And then join on on slack if you need any help: http://slack-invite.signalk.org/

coconha2, 2020-07-11:
Hi Scott, 
I just joined signalk on slack. I had some issues installing signalk on my Ubuntu PC yesterday, but I was little on time. I will give it another shot this weekend.
Thanks for the pointers!

closed 2020-07-11 by coconha2