canboat/canboat #202: n2kd does not close client sockets

wellenvogel, 2020-07-10:
When using n2kd and having an application that connects/disconnects/reconnects on port 2599 after some time n2kd becomes unusable. It consumes 100% CPU and does not accept new connections.
When stracing I see to "many open files". And with lsof I see a lot of sockets in CLOSE_WAIT.
A simple check with 
`nc localhost 2599`
and stopping the nc shows a new CLOSE_WAIT socket on n2kd afterwards.
I guess there is something wrong with the handling of closed client connections.

wellenvogel, 2020-07-10:
Ok, after some digging into the code it seems that the NMEA0183 stream misses any read handler and this way it is never checked to be readable (what will happen on close). I guess there should be a handler that at least checks for EOF and will close the socket in this case.

keesverruijt, 2020-07-11:
This is the wrong approach. Any NMEA0183 client may already send a write shutdown which translates as EOF as it does not ever intend to send data. What the code should do is do a read shutdown on the sockets that it does not intend to read from, not what your PR does which is read data that it will discard.

This is what it should do:

    shutdown(fd, SHUT_RD);

keesverruijt, 2020-07-11:
Additionally in `writeAllClients` the `select` call should check for error FDs as well as ready-to-write FDs. If it then attempts to write on the errorFD it will "see" the "socket is closed" error which it should ignore and close the socket.

wellenvogel, 2020-07-11:
Not sure if I agree with your first comment.
Until now NMEA connections are nit being read at all - so the client has no chance.
For your second proposal:
Are you sure?
I think the socket will simply never become writable (but readable) if the other end closes.
I don't think that the error set will fire in this case (as there is no error at all).
So you have to ensure to check readability.
If you feel that there is a need to really read the data - this was not there before so I thought this is intentionally.
For me any solution is fine that simply keeps n2kd operational even after many connects/disconnects.
My pr was a simple approach.

wellenvogel, 2020-07-12:
Small addition to my comment:
The only reason why I read data at all is to prevent the socket from firing again and again if the client sends any data (for whatever reason).
A read of at least one byte would be necessary to get the eof state (i.e. other end closed) in any case.

wellenvogel, 2020-07-26:
Any news on that issue? Is there a chance for a fix?

commit b7891464 (2020-08-09): Fix #202: Add read handlers on all socket streams
closed 2020-08-09 by keesverruijt