canboat/canboat #207: AIS Decoding

TwoCanPlugIn, 2020-08-13:
Hi Kees,

I think you are might be incorrectly interpreting the transceiver Information  field in AIS messages. 
I think you  have the length correct, however I don't think it is an enum, I think it is a bit map.

Bit 0 - AIS VHF Channel A or Channel B
Bit 1 - Receive or Transmit
Bit 2 - Own ship transmission

This would allow one to correctly convert to a NMEA 183 AIS sentence. 

If bit 2 is set, the NMEA 183 sentence has the prefix !AISVDO otherwise it is !AISVDM

In the NMEA 183 sentence the 4th field  indicates the AIS VHF Channel A or B.

Eg.  if bit 1 was set, indicating Channel B then a sentence would look like: 
!AIVDM,1,1,,B,177KQJ5000G?tO`K>RA1wUbN0TKH,0*5C
If bit 1 was not set, indicating Channel A the sentence would look like:
!AIVDM,1,1,,A,177KQJ5000G?tO`K>RA1wUbN0TKH,0*5C

Note that Channel A or B indicate the VHF Channels on which AIS is transmitted, and is not to be confused with AIS Class A and Class B transponders which is something entirely different. Both Class A and Class B transponders transmit and receive on AIS VHF Channels A and B.

Kind regards.

sbender9, 2020-08-13:
Can you reach out to me about this via email: scott@scottbender.net

sbender9, 2020-08-13:
I think the enum is correct. The message can only be one of these four things.

```  (",0=Channel A VDL reception"       \
   ",1=Channel B VDL reception"       \
   ",2=Channel A VDL transmission"    \
   ",3=Channel B VDL transmission"    \
   ",4=Own information not broadcast" \
```

If you were correct, that would mean that a message could be Channel B, Receive, and Own ship transmission if all bits are set

TwoCanPlugIn, 2020-08-13:
Hi Scott,

Yes, I tend to agree with you, it would be feasible to construct a Channel A or B Reception & Own Ship value. One would hope that the spec is correct although on the other hand it is quite possible that the AIS vendors have implemented it incorrectly.

Your thoughts as to how one should construct the corresponding NMEA 183 sentence ? If the value is 4, indicating an own ship message (!AIVDO) does that imply that in the NMEA 183 sentence, the AIS VHF Channel field is null ?

I've never seen a !AIVDO sentence from a NMEA 183 AIS transponder that doesn't have the AIS VHF Channel set. My understanding that when the transponder sends an AIS transmission over either the 'A' or 'B' channel, it simultaneously sends the corresponding sentence over the NMEA 183 interface.

Anyway, I'll amble along and see what breaks.

Thanks.

sbender9, 2020-08-13:
Yeah, If it's 4 then that means it was just sent on the local network, it was not broadcast on any VHF channel. That's what I see from my N2K AIS.

keesverruijt, 2020-08-16:
For a 3 bit value the last value is reserved for "null" / "unknown", as with any N2K field value (although, for >=4 bits there are 2 reserved values, "error" and "unknown").

If you need the raw values in the analyzer output to make it easier to parse you can use the `-nv` option.

closed 2020-09-30 by keesverruijt