canboat/canboat #213: Atmospheric pressure precision in bars lower than sensor spec

Vince-vd, 2020-09-28:
Hi,

I'm not 100% sure this is a canboat issue but I'm not sure where else to start.
I'm using a 220wx sensor which should be able to report atmospheric pressure with an accuracy of up to 0.1 millibar. The data produced by canboat however seems to only have an accuracy of 1 millibar. 
I have posted the data read as json and in the standard analyzer format below. In the standard format it is shown in bar, in the json format it is shown with what looks like 2 extra significant digits, but they are always zero. 

`{"timestamp":"2018-01-28T16:04:29.070Z","prio":5,"src":36,"dst":255,"pgn":130311,"description":"Environmental Parameters","fields":{"SID":15,"Temperature Source":"Outside Temperature","Humidity Source":"Outside","Temperature":27.10,"Humidity":41.600,"Atmospheric Pressure":99900}}`

`2018-01-28T16:08:49.604Z 5  36 255 130311 Environmental Parameters:  SID = 34; Temperature Source = Outside Temperature; Humidity Source = Outside; Temperature = 27.40 C (81.3 F); Humidity = 41.100 %; Atmospheric Pressure = 0.999 bar (68.9 PSI)`

Is this part of canboat's design? Or should this work and is this an issue with my sensor or actisense? I emailed airmar and they say the sensor should have no issue displaying 0.1 millibar precision.

Thank you,
Vincent

keesverruijt, 2020-09-28:
@Vince-vd can you upload some raw data before the analysis

Vince-vd, 2020-09-29:
@keesverruijt 
Sorry, of course.

Here's some data without running analyzer:

```
2018-01-28T16:20:32.172Z,5,130311,36,255,8,f7,41,13,74,ae,2e,e7,03
2018-01-28T16:20:32.173Z,5,130311,36,255,8,f8,41,13,74,ae,2e,e7,03
2018-01-28T16:20:32.174Z,5,130311,36,255,8,f9,41,13,74,ae,2e,e7,03
2018-01-28T16:20:32.175Z,5,130311,36,255,8,fa,41,13,74,ae,2e,e7,03
2018-01-28T16:20:32.436Z,5,130311,36,255,8,00,41,13,74,ae,2e,e7,03
```

This is a snippet where I used analyzer with the data flag, in case that's helpful at all. 

```
2018-01-28T16:13:01.927Z 5  36 255 130311 Environmental Parameters:  67 41 9B 73 70 30 E7 03
2018-01-28T16:13:01.927Z 5  36 255 130311 Environmental Parameters:   g  A  .  s  p  0  .  .
2018-01-28T16:13:01.927Z 5  36 255 130311 Environmental Parameters:  SID = 103; Temperature Source = Outside Temperature; Humidity Source = Outside; Temperature = 22.80 C (73.0 F); Humidity = 49.600 %; Atmospheric Pressure = 0.999 bar (68.9 PSI)
```

keesverruijt, 2020-09-29:
@Vince-vd 

PGN 130311 "Environmental parameters" has a 16 bit pressure field, with the value represented in _hectoPascal_, e.g. 1 mbar precision. So it will contain, for instance, 999 for 0.999 bar.
PGN 130314 "Actual pressure" has a 32 bit pressure field, with the value represented in _deciPascal_, e.g. 1 microbar precision, so it will contain, for instance, 1013073 for 1.013073 bar. It has a thousand times the precision of PGN 130311.

So 130311 will never do what you want.
That being said, there is a small issue in `analyzer` when used in _human_ (not JSON) mode: it will not print sufficient precision in either the Bar or PSI values to show all available data.

keesverruijt, 2020-09-29:
In JSON mode it will always print the value in Pascal, so for 32 bit it loses the last digit of precision.

Vince-vd, 2020-09-29:
@keesverruijt ah wonderful, thanks for the clarification! I should have looked more closely at 130314 myself, I did only briefly check it out in human readable format and just assumed the same thing was going on.
Having 1/10 mbar resolution is all I need for my application so that final digit in the JSON message won't be an issue for me. Bedankt voor de hulp en het snelle antwoord! :)

commit cafc250d (2020-09-29): Fix #213

Print pressure fields with the correct number of decimal digits.
PSI is a little bit inexact, of course, for Bar it should be exact.
closed 2020-09-29 by keesverruijt