canboat/canboat #225: Changing license to Apache 2.0

keesverruijt, 2021-02-08:
Given how the world has changed, and we're running into (small, perceived) licensing issues with SIgnal K, which uses CANboatJS which uses the JSON file, I want to change the license for (at least) the JSON file, but I'd rather change the license for the entire project.

The proposed new license is Apache 2.0 -- https://www.apache.org/licenses/LICENSE-2.0.html

I will add known contributors to this issue (and copies there of as I cannot add > 10 assignees). Please react if you agree to have your contributions carried over, or not. If not, please PM me and we can figure out what to do with your code.

I will not change the license before March 1, 2021 to give you time to think.

keesverruijt, 2021-02-08:
Assignees doesn't work -- let's just ping you guys?

@sbender9
@sbender9 @acolomb
@timmathews
@jpilet
@jncarter123
@chacal
@cmotelet
@saxofon
@mairas
@webmasterkai
@jonross09
@htool
@fabdrol
@e-sailing
@kledom
@cbatson
@bioffe01

acolomb, 2021-02-08:
I'm fine with the change of license.

kraj, 2021-02-08:
I am fine with changing the License to Apache-2.0

keesverruijt, 2021-02-08:
Thanks @acolomb and @kraj, 

Assigning doesn't work, pinging the remaining contributors:

@bioffe01
@cbatson
@chacal
@cmotelet
@e-sailing
@fabdrol
@francois-berder
@gobenb
@htool
@jeronimo
@jlaba
@jncarter123
@jonross09
@jpilet
@kledom
@larsujensen
@lfrancois
@lgelo
@mairas
@ph1l
@pi
@sailoog
@saxofon
@sbender9
@timmathews
@tjkurki
@tkurki
@webmasterkai
@woobagooba
@zubenubi

timmathews, 2021-02-08:
I agree to changing the license to Apache-2.0

jeronimo, 2021-02-08:
Agreed. I guess everyone who gives a 👍 agrees as well.

jonross09, 2021-02-08:
👍

sailoog, 2021-02-08:
Agree

jpilet, 2021-02-08:
Approved, very nice change.

tkurki, 2021-02-08:
Approved 👍

kledom, 2021-02-08:
👍

tjkurki, 2021-02-08:
Approved - why on earth did I ever use a separate account for something here, as this is me Teppo also.

lgelo, 2021-02-08:
👍 Agree.

htool, 2021-02-08:
Agree 👍

cbatson, 2021-02-09:
Works for me 👍 Presumably the license change cannot be retroactive? I.e. earlier revisions are still subject to the former license?

sbender9, 2021-02-09:
@cbatson I think you're right, but I don't think it matters if it's retroactive in this case. Only if someone here would try to enforce the GPL on people using older versions. Clearly that is not the intention going forward.

mairas, 2021-02-09:
This is a great change! Wholeheartedly agree.

saxofon, 2021-02-09:
OK from my side too.

e-sailing, 2021-02-09:
👍 Agree.

cmotelet, 2021-02-09:
👍 Agree.

jlaba, 2021-02-09:
Agree as well

multimike, 2021-02-12:
Agreed

ph1l, 2021-02-14:
I'm confused..

So is there an example of the "small, perceived" licensing issues?

This changes the license from an open source copyleft license, which preserves our right to benefit from others (including corporations) modifying the code in the future, to a "do whatever you want" and profit off of the project without requiring to give back to the community license?

And this change, to the entire codebase, fixes one issue with one file used by one project?

:-1:

sbender9, 2021-02-15:
@ph1l this decision was not about one project using one file. Where did you get that impression?

ph1l, 2021-02-15:
> I want to change the license for (at least) the JSON file

sbender9, 2021-02-15:
Yes, that was how this discussion got started, but it goes much deeper than that. The GPL (arguably) would not effect anyone using that file anyway since it is not "linked" into any applications. IMHO the GPL is not relevant anymore and it's not policeable or enforceable by small open source projects like ours. We're not going to be suing anyone over this stuff, that costs money.

There are small companies using this stuff, it's not the big guys. I think that's a good thing and I'd prefer that they didn't have worry about these legal issues. 

I want entrepreneurs to be able to innovate, create great products, and employ people.

sbender9, 2021-02-15:
And you say "preserves our right to benefit from others"? You have three tiny commits. Good luck with that.

keesverruijt, 2021-02-15:
Hi Phil, this is not a discussion on whether this license will be applied or not. I have carefully weighed the disadvantages and advantages.

The single file you talk about is the real meat of the project: the reverse-engineered NMEA and private PGN layout. The rest is totally boring C code that can easily be rewritten from scratch. 

The people we are restraining with the license are the small fledgling businesses that are interested in NMEA2000 and would like to get started with a small leg up before they invest in the NMEA membership and license fees. I have no issue with them starting with our code. They have to move up to NMEA membership if they want to certify NMEA 2000 compatibility anyway.

So the decision was made to go forward with this, to great acclaim from all contributors so far. If you really don't want to, I will revert your 3 commits (that do not involve the PGN list anyway.)

keesverruijt, 2021-02-15:
Note that the other negative vote on the initial post in this issue from @vcaputo is from a non-contributor who collaborated on a different project with @ph1l ( https://github.com/vcaputo/rototiller ).

vcaputo, 2021-02-15:
> Note that the other negative vote on the initial post in this issue from @vcaputo is from a non-contributor who collaborated on a different project with @ph1l ( https://github.com/vcaputo/rototiller ).

Perhaps I'm misreading, that comes across as negative and conspiratorial, which is surprising.  But then again, this is basically the twitter of software development.

@ph1l  and I are friends and frequent Libre-software collaborators.

I'm not a copyright holder of any code in this project, but @ph1l and I have discussed these subjects at length when sailing or reviewing/testing code outside of github (e-mail, IRC) over the years.  We've collaborated on more than just rototiller.  I have no legal basis to argue license changes here, but socially speaking, I do have an opinion.  And this being github, you're essentially inviting it.

keesverruijt, 2021-02-15:
> Perhaps I'm misreading, that comes across as negative and conspiratorial, which is surprising. But then again, this is basically the twitter of software development.

Sorry? Github like Twitter? I guess your opinion is clear...

> I'm not a copyright holder of any code in this project, but @ph1l and I have discussed these subjects at length when sailing or reviewing/testing code outside of github (e-mail, IRC) over the years. [...]

I am more than happy to have you chime in, especially when this would be a discussion on the merits of different software licenses, but it is not. The die has been cast on that; the license _will_ change. I am asking collaborators whether they want their contributions to roll over into the Apache-licensed next version. If they don't, fine, it will be removed and, if essential, re-engineered by someone else.

> And this being github, you're essentially inviting it.

I asked _contributors_ to chime in whether they vote yes/no, and I also asked contributors to PM me with what I should do with their source if they do not agree to their contributions being Apache-licensed. My comment was intended to point out that we can disregard your negative vote as you are not a contributor. Sorry if that came across as negative. It certainly wasn't "conspirational". I also haven't heard from @phil what I should do with his contributions.

phil, 2021-02-17:
I have no idea; and TBH, I'm probably not the right person to ask … maybe try @ph1l ?

commit 64fe6c81 (2021-03-10): Fix #225: Change license to Apache License 2.0

Bumps version to 2.0.0 because of license change.
"
closed 2021-03-10 by keesverruijt