canboat/canboat #235: Incorrect generated schema in pgns.json

aldas, 2021-05-18:
I'm trying to unmarshal pngs.json in Go I think somewhere during the XML2json creation process there was a mistake and two PGNs have incorrect fields definition

There are PGNs json objects that have `Fields` key as object containg key `Field` but most of the `Fields` are arrays containing object. Seems to be incorrectly generated json. resembles more like XML structure that way.

https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgns.json#L69

https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgns.json#L4231

https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgns.json#L7183

aldas, 2021-05-18:
Another note. `Resolution` fields values are sometimes json numbers and sometimes strings

`"Resolution":1,`
`"Resolution":"0.0001",`

JS does not care much about it but for stricter languages it is a problem. But even for JS  string vs number could lead to unexpected problems.

tkurki, 2021-05-18:
Uuh, it's been a while and my xslt is rusted over, bad.

But I think the problem is that outputting `Field` s  in arrays is triggered by having sibling elements: https://github.com/canboat/canboat/blob/master/analyzer/pgns2json.xslt#L315-L328 

If I am not mistaken creating an explicit rule to produce an array for each `Fields` element would be one way to fix this.

Fix for numbers in #236.

aldas, 2021-05-24:
I know - do you guys mind if I keep posting remarks here?

Anyway - there 3 Windlass PGN fields that are generated incorrectly because their type RES_BITFIELD would expect `description` value to start with `,` to be correctly parsed into `EnumBitValues`. At the moment json does not have `EnumBitValues` even if type says `Bitfield`

https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgn.h#L3393
https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgn.h#L3418
https://github.com/canboat/canboat/blob/732371ada8b0c6f33652c3ab61f0856abfd9e076/analyzer/pgn.h#L3436

PR #242 (2021-07-28): fix #235: Incorrectly generated schema in pgns.json for Fields.Field arrays
fix #235: Incorrectly generated schema in pgns.json for Fields.Field values, and also fix Windlass enum values
commit 227bf4f4 (2021-07-28): Merge pull request #242 from aldas/fix_fields_array

fix #235: Incorrectly generated schema in pgns.json for Fields.Field arrays
commit 23d3aaaf (2021-07-28): fix #235: Incorrectly generated schema in pgns.json for Fields.Field values,  Windlass enum values
commit 53b25aa9 (2021-07-29): Merge pull request #1 from aldas/fix_fields_array

fix #235: Incorrectly generated schema in pgns.json
closed 2021-07-28 by keesverruijt