canboat/canboat #243: PGN127507 Defined as a single frame, is actually a multi-frame

nmbath, 2021-07-13:
I believe that PGN127507 is a multi-frame packet. When I compare what Actisense NMEA reader shows for 127507 and canboat/canboatjs there is a distinct difference.

Note that Actisense shows 6 data bytes in its decode.

When I decode using canboat/canboatjs it uses all 8 bytes and the charger instance is all over the place.

[image]

keesverruijt, 2021-07-28:
I see that your source shows that, what device creates that data?

I have a different source that generates this:

    2020-08-22T13:52:39.655Z,7,127507,1,255,8,00,01,f0,fc,ff,ff,01,00
    2020-08-22T13:52:41.109Z,7,127507,1,255,8,00,01,f0,fc,ff,ff,00,00

so I need a little more "proof". A longer log from your can device would do.

nmbath, 2021-07-28:
I hope the following traces and details help.

I have two devices, a Wakespeed controller which has three charging sources inside it, and generates device instances 0x31 (49), 0x71 (113) and 0x72 (114), these are all on CAN address 0x81(129). I also have a a Victron Cerbo GX, which is putting data into the network from CAN address 0xE1 (225), which is a MultiPlus inverter.

The following CAN traces is a repeating set of 127507 PGNs:

 (2021-07-28 13:21:18.097254)  can1  RX - -  19F213E1   [8]  00 06 00 FF F8 FD FF FF
 (2021-07-28 13:21:18.326661)  can0  RX - -  19F21381   [8]  E0 06 31 00 08 FC FF FF
 (2021-07-28 13:21:18.340841)  can0  RX - -  19F21381   [8]  00 06 71 00 08 FC FF FF
 (2021-07-28 13:21:18.342958)  can0  RX - -  19F21381   [8]  20 06 72 06 08 FC FF FF

What has lead me to believe that this is a fast packet, is that all the devices instances are in the 3rd byte. The 4th Byte then aligns with the battery instance. Everything except charging instance 0x72 (114), is associated with a battery instance 0, while 0x72 is associated with battery instance 6.

Based on this it would seem that the first byte is the fast packet, with the second byte being the length. the value 0x06 aligns with the length in the Actisense screen shot.

If this was not a fast packet then the length value from Actisense would not make sense. It would also mean that the Charger instances would be 0x00 (0), 0xE0 (224) and 0x20 (32). These do not align with the Wakespeed documentation, or the Victron configurations and details.

It would also mean that the second byte (0x06) would be the battery instance, and I certainly know that not all four charges are associated with the same battery instance.

I attach a much longer candump, and candumpanalyzerjs output. As you will see from these outputs the Charger Instance IDs would be 0, 192, 32, 64, 96, 224, 128 and 160. 

If we then apply the above to your messages, there is a very different outcome,

00,01,f0,fc,ff,ff,00,00

The 3rd byte for charger instance would be 0xF0 (240) and battery instance would be 0xFC (252)

What device do you have? It would seems strange that with the PGN decoded as a fast packet, decode of Charger Instance and Battery Instance would be correct, and when used as a single packet the 

[PGN127507.zip](https://github.com/canboat/canboat/files/6894890/PGN127507.zip)

dbeckwith01, 2021-07-28:
It's interesting that in the Actisense screen shot it says DEPRECATED.  According to [this](https://www.victronenergy.com/upload/documents/Technical-Information-Data-communication-with-Victron-Energy-products_EN.pdf) 127507 is deprecated for 127750 but can still be requested and set to transit on an interval.

I agree with @nmbath it does look like 127507 is a Fast PGN, here is my data:
2021-01-30-20:39:57.151,6,127507,225,255,8,c0,06,00,ff,f2,fd,ff,ff
2021-01-30-20:39:58.641,6,127507,225,255,8,e0,06,00,ff,f2,fd,ff,ff
2021-01-30-20:40:00.145,6,127507,225,255,8,00,06,00,ff,f2,fd,ff,ff
2021-01-30-20:40:01.638,6,127507,225,255,8,20,06,00,ff,f2,fd,ff,ff

nmbath, 2021-07-28:
Interestingly, I have modified pgn.h and pgns.json to set 127507 to be type FAST, however I think there maybe more to it.

When I pass one of the PGNs candumpanalyzerjs it stills parses it incorrectly.

  canboatjs:fromPgn parsed pgn {"canId":435295105,"prio":6,"src":129,"dst":255,"pgn":127507,"timestamp":"2021-07-28T19:25:11.582Z","fields":{"Instance":160,"Battery Instance":6,"Operating State":"Off","Charge Mode":7,"Equalization Pending":"Off","Reserved1":"0","Equalization Time Remaining":64520},"description":"Charger Status"} +0ms

For some reason it isn't seeing it as a fast packet. It looks like the the function _parse within fromPgn.js is not able to detect a fast packet when the length is < 8 bytes.  There are a few pins defined in pins.json that are fast and have a length of < 8, but none that I have example can messages, which I could use to test further.

dbeckwith01, 2021-07-29:
@nmbath I pulled the latest main branch and updated 127507 in pgn.h to be FAST and it seems to be correct for me.  Here is the output of analyzer using the zip file you provided above.  See below.  Could this be a problem with candumpjs?

```
INFO 2021-07-29T04:18:47.875Z [analyzer] Assuming normal format with one line per packet
INFO 2021-07-29T04:18:47.875Z [analyzer] New PGN 127507 for device 225 (heap 11151 bytes)
2021-07-29-04:18:26.112 6 225 255 127507 Charger Status:  Instance = 0; Battery Instance = Unknown; Operating State = Disabled; Charge Mode = Unknown; Operating State = On; Equalization Pending = Unknown; Equalization Time Remaining = Unknown
INFO 2021-07-29T04:18:47.875Z [analyzer] New PGN 127507 for device 129 (heap 22302 bytes)
2021-07-29-04:18:26.112 6 129 255 127507 Charger Status:  Instance = 49; Battery Instance = 0; Operating State = Disabled; Charge Mode = Standalone mode; Operating State = Off; Equalization Pending = Unknown; Equalization Time Remaining = Unknown
2021-07-29-04:18:26.112 6 129 255 127507 Charger Status:  Instance = 113; Battery Instance = 0; Operating State = Disabled; Charge Mode = Standalone mode; Operating State = Off; Equalization Pending = Unknown; Equalization Time Remaining = Unknown
2021-07-29-04:18:26.112 6 129 255 127507 Charger Status:  Instance = 114; Battery Instance = 6; Operating State = Disabled; Charge Mode = Standalone mode; Operating State = Off; Equalization Pending = Unknown; Equalization Time Remaining = Unknown
2021-07-29-04:18:26.112 6 225 255 127507 Charger Status:  Instance = 0; Battery Instance = Unknown; Operating State = Disabled; Charge Mode = Unknown; Operating State = On; Equalization Pending = Unknown; Equalization Time Remaining = Unknown
```

In data I provided and some of the lines in the data your provided the battery instance marked as FF, do you know why?

nmbath, 2021-07-29:
@dbeckwith01 I should have said that I had been testing with canboatjs, as did not have canboat installed.  Will download canboat and test. Interesting some have battery instance as FF. I have scanned through the candump file and do see the messages from device 0xE1 (225) the Victron MultiPlus, has a battery instance of FF. I guess Victron has done this as they do not have detail of what the NMNEA200 battery instance is that the MultiPlus is connected too.

nmbath, 2021-07-29:
@dbeckwith01, Just pulled down canboat, modified the PGNS to fast, compiled it and run a test and can confirm I get the correct output. Will raise an issue on canboatjs.

keesverruijt, 2021-07-29:
Thanks guys (assuming!) this is really helpful. For years these PGNs haven't been seeing much use apparently so nobody saw these issues. 

My original sample is from a friend's boat, I am not sure what device is producing the logs I had.

The fact that a Victron is producing these has me convinced, they are rigorous, so I'm pushing a fix soon.

closed 2021-07-29 by keesverruijt

nmbath, 2021-07-29:
My pleasure and glad I could help. As I see further issues I will raise separate issue. I am starting to monitor my electrical systems through SignalK and storing data into an influxDB.....

dbeckwith01, 2021-07-30:
@keesverruijt It look like this merge d997bafc5e892e65c5ece1051d9fa8983ddce8fc removed the fix for this.

commit 71615ffd (2021-07-29): Fix #243: PGN 127507 is fast packet