canboat/canboat #256: PGN 129208 "Altitude Delta, Rapid Update" flows over declared length

mairas, 2021-09-21:
PGN 129208 defines a length of 8 but there are 10 bytes worth of fields in the definition. Should the type be different?

keesverruijt, 2021-09-28:
I think the most likely error in the definition is that the Course over Ground should be two bytes, not four. This is the case for all other COG fields. I think the name should also be changed to "COG".

keesverruijt, 2021-09-28:
Although I don't understand what COG is doing in an "Altitude delta" PGN. 

Do you have a sample of these on a live bus?

mairas, 2021-09-29:
> Do you have a sample of these on a live bus?

Unfortunately, not. I just bumped into this when writing the DBC exporter and the DBC message exporter complained about the signals overflowing the declared message length (paraphrasing, in DBC lingo). I was thinking that either there's something funky with the field definitions, or then it should be a fast packet. But given that it's a rapid update message, a fast packet can be ruled out.

[This document](https://www.nmea.org/Assets/july%202010%20nmea2000_v1-301_app_b_pgn_field_list.pdf) states the accuracy of COG is as small as 0.0057 degrees which corresponds to resolution of 0.0001. And 2*pi does fit in 16 bits at that resolution, so that's got to be it, just like you suspected.

Digressing, but I guess it makes sense to have the COG present in that PGN. This way, the PGN gives you an atomic location delta in 3D space, which, I suppose, would come handy if you're providing instrumentation for the Flying Dutchman.

commit 086fb899 (2021-10-04): Fix #256: Fix COG in PGN 129028
closed 2021-10-04 by keesverruijt